openshwprojects/OpenBK7231T_App #499: MQTT work good. Not send ADC channel to MQTT (help)

alexxx113, 2022-11-25:
https://github.com/openshwprojects/OpenBK7231T_App/issues/116
ok, no bug !!!

Flag 2 - [MQTT] Broadcast self state every minute (May cause device disconnect's, DONT USE IT YET)
no actual: May cause device disconnect's, DONT USE IT YET

how to get ADC ?

before work:

configure modul
P23 ADC 5

general config
Flag 2

Home Assistant Cfg
  - unique_id: mqtt_temp_sens_koridor
    state_topic: "koridor/5/get"
    name: "mqtt temp sens koridor"
    qos: 1
    device_class: power
    availability:
      - topic: "koridor/connected"

now doesn't work ((((

openshwprojects, 2022-11-25:
I know what's wrong. It seems we don't check for ADC role. Please wait, I will check.

alexxx113, 2022-11-25:
Thank you, I'm waiting!!!

openshwprojects, 2022-11-25:
ADC channel was skippde while doing publish all! Sorry. Can you try now?
https://github.com/openshwprojects/OpenBK7231T_App/commit/8c858f8492ece1ed1a72d3ee100042cbbb29dfef

alexxx113, 2022-11-25:
Thank you !!!!!
Work good!!!!

closed 2022-11-27 by openshwprojects