openshwprojects/OpenBK7231T_App #502: Calibration settings not saved initially

ben-willmore, 2022-11-26:
When I first tried to calibrate my devices, the VoltageSet, CurrentSet and PowerSet commands did nothing -- I got the "Command not found" error (#501) and nothing else happened. Strangely, after I set up long and short MQTT names, the commands started working correctly (though I still get the Command not found error).

* Version 1.15.70/1.15.78
* LSPA7 https://www.elektroda.com/rtvforum/viewtopic.php?p=20300613#20300613
* Chip/model: BK7231T

openshwprojects, 2022-11-26:
I think you've found two issues - one is a bug that makes "Command not found" print while command is executed, and second is not a bug, but... but commands started working correctly because the driver for BL0937 starts after reboot and most likely you have rebooted?
Wait, I will fix the "command not found" first

ben-willmore, 2022-11-27:
I don't think it's about the BL0937 driver being loaded -- it seems like it always gets loaded automatically. I'm testing now with 1.15.90, and VoltageSet now works if I have `startDriver BL0937` as a Startup Command. If I set the Startup Command to blank and reboot, the BL0937 driver is still loaded, and VoltageSet still works.

openshwprojects, 2022-11-27:
@ben-willmore this is because, if on startup, you have BL0937 pins set, it will start automatically.

If you set pins for BL0937, but don't restart, then driver is not started yet.

This is why for the first time, just once, you get "unknown commands"

ben-willmore, 2022-11-27:
OK, makes sense. I couldn't reproduce this on a cleared device, so I will close this.

ben-willmore, 2022-11-27:
I went through this again with my other plugs (same model), after upgrading to 1.15.90. Now, VoltageSet works, but the value is not saved across reboots. 

1. Check voltage in main page -- 242V
[image]

2. Calibrate voltage to 100V
[image]

3. Check voltage again -- it's correctly changed to 100V
[image]

4. Reboot the device and check voltage again -- it's back to 240V
[image]

I cleared the configuration on this device and set it up as new, and got the same result.

ben-willmore, 2022-11-27:
I worked out why this seemed intermittent to me. If I hit "Restart" in the GUI after running VoltageSet, then the value is correctly saved, and I can then cut power to the device and it will restart with the correct calibration. However, if I cut power to the device straight after VoltageSet, the calibration is lost when the device restarts.

openshwprojects, 2022-11-27:
ah, now I sese what you mean, thanks. There was an obvious bug, the save flush was missing for VoltageSet etc but flush was happening if you changed something else later so people didn't notice, most of the time...
https://github.com/openshwprojects/OpenBK7231T_App/commit/90f68934fa300d524e8549713d5f7d1bbe976f25
next release should have it fixed

ben-willmore, 2022-11-27:
So quick, thanks!

closed 2022-11-27 by ben-willmore

openshwprojects, 2022-11-27:
If there are any other strange issues, let me know