openshwprojects/OpenBK7231T_App #503: Device Group - Colors not synced with other openbk lights

ferbulous, 2022-11-26:
**Describe the bug**
Color/color temperature not synced with other openbk lights.
Presently, this only works if I were to change colors from tasmota lights, or manually send dgr commands

**Firmware:**
- Version : 1.14.78
- Device : Light Bulb
- Chip/model: BK7231N
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to openbk ui/ha dashboard
2. Change the colors/color temperature for 1 openbk light
3. Other openbk lights in the same device group colors is not changed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Log output when color is changed
Bulb 1
```
Info:MAIN:Time 395, idle 453193/s, free 76488, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 396, idle 451294/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 397, idle 450188/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:DGR:DGR_Parse: grp name test len 4
Info:DGR:DGR_Parse: grp name nightlamp len 9
Info:DGR:DGR_Parse: grp name nightlamp len 9
Info:MAIN:Time 399, idle 453802/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 400, idle 903841/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-44,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=11,cipher_type:CCMP
Info:MAIN:Time 402, idle 435314/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 403, idle 452597/s, free 68088, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 412, idle 455803/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x001B, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:MAIN:Time 416, idle 454096/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x001D, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:MQTT:Publishing val 1 to smartrul_bulb5/led_enableAll/get retain=1
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:MAIN:Time 418, idle 456622/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x001F, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x0020, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:MAIN:Time 419, idle 439982/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 420, idle 437583/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-45,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=11,cipher_type:CCMP
Info:MAIN:Time 422, idle 447467/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 424, idle 452649/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 425, idle 452755/s, free 76704, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
```

Bulb 2
```
[... 53 lines cut ...]
Info:MQTT:Publishing val 0000007200 to aubess_bulb1/led_finalcolor_rgbcw/get retain=1
Info:MQTT:Publishing val 1 to aubess_bulb1/led_enableAll/get retain=1
Info:MAIN:Time 4445, idle 424741/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x001E, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:MAIN:Time 4446, idle 426548/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 4447, idle 447190/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x001F, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:DGR:DGR_Parse: grp name cornerlamp len 10
Info:DGR:DGR_Parse: [192.168.1.239] seq 0x0020, flags 0x00
Info:CFG:####### Flash Save LED #######
Info:MAIN:Time 4448, idle 415465/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 4449, idle 443102/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 4450, idle 449596/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:GEN:dhcp=0 ip=192.168.1.130 gate=192.168.1.1 mask=255.255.255.0 mac=fc:67:1f:59:cf:a2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-51,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=11,cipher_type:CCMP
Info:MAIN:Time 4451, idle 437912/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 4452, idle 440837/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 4453, idle 444579/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
Info:MAIN:Time 4454, idle 446332/s, free 75856, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 6/38
```

openshwprojects, 2022-11-26:
Wow thanks, I totally forget about that one. 
I am unable to test right now because I am running long term stability experiment on my bulbs, but can you check this commit?
https://github.com/openshwprojects/OpenBK7231T_App/commit/cea9364ddc4e6bd41e14999326a75030ac07f718
Build is: 1.15.82

Please check and report here if it works as expected, it's just my shot in the dark, without testing so far.

ferbulous, 2022-11-26:
@openshwprojects thanks! that has fixed this issue
will continue monitoring for issue #477 for 1.15.82

ferbulous, 2022-11-27:
@openshwprojects I'm noticing some odd issue when I'm simply toggling on/off the lights.
Some of the lights reverting to this black color (basically no lights at all), and I need to change the color/color temperature again to have it light up.

```
_enableAll/get retain=1
Info:DGR:Ignoring message from self
Info:DGR:Ignoring message from self
Info:MAIN:Time 33706, idle 449495/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33707, idle 475446/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33708, idle 476254/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33709, idle 475121/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33710, idle 473552/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-58,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=10,cipher_type:CCMP
Info:MAIN:Time 33711, idle 476210/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:DGR:DGR_Parse: grp name nothing len 7
Info:MAIN:Time 33712, idle 475267/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33713, idle 474525/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33714, idle 470902/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33715, idle 475912/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33716, idle 473648/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33717, idle 473604/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33718, idle 476810/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33719, idle 953761/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33720, idle 472614/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-58,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=10,cipher_type:CCMP
Info:MAIN:Time 33721, idle 473578/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33722, idle 475728/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33723, idle 475810/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33724, idle 472002/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33725, idle 473468/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33726, idle 469493/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33727, idle 471084/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33728, idle 472666/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33729, idle 458808/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33730, idle 468329/s, free 62280, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 19/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-58,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=10,cipher_type:CCMP
Info:MAIN:Time 33731, idle 471207/s, free 62280, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 19/38
Info:MAIN:Time 33732, idle 465026/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33733, idle 473568/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33734, idle 464525/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:DGR:DGR_Parse: grp name nightlamp len 9
Info:MAIN:Time 33735, idle 458403/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33736, idle 465975/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33737, idle 442850/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33738, idle 449179/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33739, idle 450863/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33740, idle 465552/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-58,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=10,cipher_type:CCMP
Info:MAIN:Time 33741, idle 461667/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33742, idle 466880/s, free 73760, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 19/38
Info:MAIN:Time 33743, idle 457205/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33744, idle 458342/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:MAIN:Time 33745, idle 459843/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
[... cut ...]
Info:MAIN:Time 33840, idle 453507/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-58,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=10,cipher_type:CCMP
Info:MAIN:Time 33841, idle 451953/s, free 73976, MQTT 1(74), bWifi 1, secondsWithNoPing -1, socks 18/38
```

[image]

https://user-images.githubusercontent.com/24526737/204119980-919304c9-3765-4a7d-955c-fd55f8d1ddd0.mp4

ferbulous, 2022-11-30:
With 1.15.118
The 'black light' is still there.
Other issues observed:
Corner switch, obk and aubess are in the same device group
aubess does receive dgr commands, but doesn't retransmit them?
also the brightness sliders keeps moving? not sure if this device group related or openbk light bulbs bug

https://user-images.githubusercontent.com/24526737/204857786-a70dee2a-31f9-4437-b58b-def4f3fa1654.mp4

https://user-images.githubusercontent.com/24526737/204859916-9541ae45-7eef-40b7-b9f7-401c8dc45467.mp4