openshwprojects/OpenBK7231T_App #504: CAnnot flash Merkury smart plug (US)

schneemaier, 2022-11-27:
**Describe the bug**
I have 2 Merkury Smart plug (US) with WB2S modul. After peingin the house i was able ot connect to the serial port for flashing, but it does not work. From time to time I am able to backup the content of the flash, but I am never able to write it. 

This is the error i get for reading
```
pi@raspberrypi:~/hid_download_py $ ./uartprogram -d /dev/ttyUSB0 -r gosundbk7231.bin
UartDownloader....
Read Getting Bus...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 537, in write
    n = os.write(self.fd, d)
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./uartprogram", line 56, in <module>
    downloader.read(args.filename, args.startaddr, args.length)
  File "/home/pi/hid_download_py/bkutils/uart_downloader.py", line 78, in read
    r = self.bootItf.LinkCheck()
  File "/home/pi/hid_download_py/bkutils/boot_intf.py", line 106, in LinkCheck
    rxbuf = self.Start_Cmd(txbuf, CalcRxLength_LinkCheck(), 0.001)
  File "/home/pi/hid_download_py/bkutils/boot_intf.py", line 37, in Start_Cmd
    self.ser.write(txbuf)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 571, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 19] No such device
```
and for writing
```
pi@raspberrypi:~/hid_download_py $ ./uartprogram -d /dev/ttyUSB0 -w OpenBK7231T_UA_1.15.70.bin
UartDownloader....
programm....
Getting Bus...  : |                                                  |[    ?k/s]Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 537, in write
    n = os.write(self.fd, d)
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./uartprogram", line 59, in <module>
    downloader.programm(args.filename, args.startaddr)
  File "/home/pi/hid_download_py/bkutils/uart_downloader.py", line 193, in programm
    r = self.bootItf.LinkCheck()
  File "/home/pi/hid_download_py/bkutils/boot_intf.py", line 106, in LinkCheck
    rxbuf = self.Start_Cmd(txbuf, CalcRxLength_LinkCheck(), 0.001)
  File "/home/pi/hid_download_py/bkutils/boot_intf.py", line 37, in Start_Cmd
    self.ser.write(txbuf)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 571, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 19] No such device
```

I am using the 3.3V from the raspberry to power the plug and with that power I am able to operate it (ex pressing the button triggers the relay etc). Could it be that the power is not stable enough to do the flashing? 

thanks

openshwprojects, 2022-11-27:
Is /dev/ttyUSB0 a valid device path?

schneemaier, 2022-11-27:
Yes /dev/ttyUSB0 is a valid path. I use an USB to serial adapter. I was already able to flash a few Gosund switches with the same methode

openshwprojects, 2022-11-27:
@schneemaier is the bin path correct?

schneemaier, 2022-11-27:
Yes it is correct. It worked well with the Gosund switches and as I mentioned in the first post: from time to time I am able to execute a full flash backup for these plugs, but not able to write. So I guess the serial setup is correct too.

schneemaier, 2022-11-27:
Okay, i tried it with another 3.3V power source and now it worked. I was able to flash the chip and now it is running this firmware.

Thank you for your work on this firmware!

closed 2022-11-27 by schneemaier

openshwprojects, 2022-11-27:
Good job. If your device is not listed here: https://openbekeniot.github.io/webapp/devicesList.html
please consider submitting a teardown. Thanks!

schneemaier, 2022-12-08:
Here is the teardown: https://www.elektroda.com/rtvforum/viewtopic.php?p=20324233#20324233