openshwprojects/OpenBK7231T_App #505: Missing command to delete/edit/list eventhandlers (feature request)

Naguz, 2022-11-27:
**Describe the bug**
There is no documented command to list or remove event handlers (https://github.com/openshwprojects/OpenBK7231T_App#console-commands) and I can't find any in the codebase either. After experimenting a bit with different event handlers/scripts etc to get a good setup, this means you need to reset the device and start again with all the working commands you hopefully noted down somewhere.

**Firmware:**
- Version 1.15.94

**To Reproduce**
Add a lot of event handlers. Some with errors. Try to remove some.

**Screenshots**
N/A

openshwprojects, 2022-11-28:
Thanks, should be solved now, but I will need to also update readme.

Naguz, 2022-11-29:
Thanks, that was fast! 

It would be very nice to also be able to list all event handlers (and change handlers for that matter) and delete just some of them.

openshwprojects, 2022-11-29:
Hmm so basically add an optional ID param to event handler and then allow to remove by ID?

Naguz, 2022-11-29:
Yes, and/or ideally add a random numerical ID so you can always remove an event handler added by mistake, even if you didn't manually specify an ID. I don't think I would care that much if I could set the ID it myself or not.

openshwprojects, 2022-11-29:
I think that ID should be given in command, because it would be also useful for scripting.

Naguz, 2022-11-30:
Yes, it is probably a good idea. But perhaps also set a random ID if none is provided, so you can identify/delete it later.