openshwprojects/OpenBK7231T_App #507: Feature request: MQTT retain flag

ben-willmore, 2022-11-29:
My OpenBK devices are running smoothly now, which is great.

One issue is that the relay on/off state in home assistant is often wrong, because the relevant MQTT messages are sent without the retain flag.

It would be cool if I could choose to set the retain flag on power state messages. It [looks](https://tasmota.github.io/docs/MQTT/#command-flow) like tasmota does this with a command "PowerRetain".

openshwprojects, 2022-11-29:
Hello, do you mean this?
[image]

ben-willmore, 2022-11-29:
No -- if all published values have the retain flag set, then the current/voltage/power readings would also get retained, which would lead to problem with power measurement. The only value (for my setup) that should be retained is the state of the relay. Tasmota actually has two commands, PowerRetain and SensorRetain which set the retain flag separately for power and sensor messages.

openshwprojects, 2022-11-29:
So basically you need a flag.... to enable retain, but only on power channels?

ben-willmore, 2022-11-29:
Yes, exactly.

openshwprojects, 2022-11-29:
Ok I pushed that flag with last commit ,can you check if it works OK? I am unable to fully check it right now.

ben-willmore, 2022-11-30:
Yes, this works great for me. Thanks! The only issue is that after the OTA update, when the module restarted and I toggled the button, no MQTT messages were sent at all, so the status was initially broken. It worked fine when I set the flag and rebooted again.

openshwprojects, 2022-11-30:
Sorry, can you explain again?

Flags are saved when set. They should be remembered after reboot.

Or did you toggle button before WiFi was up?

Do you have "broadcast full state on MQTT connect" flag set?

ben-willmore, 2022-11-30:
Here is the whole process I went through (on another module, same result):
```
Check MQTT Works in HA
-> confirmed, relay status is ON

Run OTA upate from web app
-> wait for sequence to finish

Check local GUI -- wifi is up and module has rebooted (uptime of a few seconds)

Reload local and web UIs

Check HA
-> relay status is unknown

Toggle relay button in local GUI
-> no updates in HA -- relay status still unknown
-> OpenBK web app has no log messages re MQTT
**UNEXPECTED -- MQTT messages should be working as they did before the update**

Set flag 21 in local GUI and press submit. No other flags are set.

Toggle relay button in local GUI
-> No updates in HA -- relay status still unknown
-> OpenBK web app has no log messages re MQTT

Reboot OpenBK
-> HA has correct MQTT status
-> OpenBK web app has correct log messages re MQTT
-> Retain flag works correctly

Toggle relay button in local GUI
-> HA updates correctly
-> OpenBK web app has correct log messages re MQTT
-> Retain flag works correctly
```

openshwprojects, 2022-11-30:
Just to be sure, is the device functional after  that? Or does the problem repeats every time?

ben-willmore, 2022-11-30:
Yes, it works fine now, after a couple of hours and reboots. Maybe I just needed to reboot again after the OTA update.

openshwprojects, 2022-12-01:
@ben-willmore how it's going after one day, is it working correctly?

I am not sure if we can close this issue.

ben-willmore, 2022-12-01:
It is 100% working for me. If the "**UNEXPECTED**" line above is actually expected, then you should close this :-)

closed 2022-12-05 by ben-willmore