openshwprojects/OpenBK7231T_App #508: w600 boot looping

mikimause9, 2022-11-29:
Hi!
I found in my trash the w600 module which I replaced a while ago with esp8266. Because I found this project I wanted to reprogram this module. The only problem is that right now the module is in some sort of boot loop. The serial output looks like this:
`w600 Compiled @Nov 15 2022 14:45:49
delaying start
#Startup delayed 0ms#
#Startup delayed 10ms#
#Startup delayed 20ms#
#Startup delayed 30ms#
#Startup delayed 40ms#
#Startup delayed 50ms#
#Startup delayed 60ms#
#Startup delayed 70ms#
#Startup delayed 80ms#
#Startup delayed 90ms#
#Startup delayed 100ms#
#Startup delayed 110ms#
#Startup delayed 120ms#
#Startup delayed 130ms#
#Startup delayed 140ms#
#Startup delayed 150ms#
#Startup delayed 160ms#
#Startup delayed 170ms#
#Startup delayed 180ms#
#Startup delayed 190ms#
#Startup delayed 200ms#
#Startup delayed 210ms#
#Startup delayed 220ms#
#Startup delayed 230ms#
#Startup delayed 240ms#
#Startup delayed 250ms#
#Startup delayed 260ms#
#Startup delayed 270ms#
#Startup delayed 280ms#
#Startup delayed 290ms#
#Startup delayed 300ms#
#Startup delayed 310ms#
#Startup delayed 320ms#
#Startup delayed 330ms#
#Startup delayed 340ms#
#Startup delayed 350ms#
#Startup delayed 360ms#
#Startup delayed 370ms#
#Startup delayed 380ms#
#Startup delayed 390ms#
#Startup delayed 400ms#
#Startup delayed 410ms#
#Startup delayed 420ms#
#Startup delayed 430ms#
#Startup delayed 440ms#
#Startup delayed 450ms#
#Startup delayed 460ms#
#Startup delayed 470ms#
#Startup delayed 480ms#
#Startup delayed 490ms#
#Startup delayed 500ms#
#Startup delayed 510ms#
#Startup delayed 520ms#
#Startup delayed 530ms#
#Startup delayed 540ms#
#Startup delayed 550ms#
#Startup delayed 560ms#
#Startup delayed 570ms#
#Startup delayed 580ms#
#Startup delayed 590ms#
#Startup delayed 600ms#
#Startup delayed 610ms#
#Startup delayed 620ms#
#Startup delayed 630ms#
#Startup delayed 640ms#
#Startup delayed 650ms#
#Startup delayed 660ms#
#Startup delayed 670ms#
#Startup delayed 680ms#
#Startup delayed 690ms#
#Startup delayed 700ms#
#Startup delayed 710ms#
#Startup delayed 720ms#
#Startup delayed 730ms#
#Startup delayed 740ms#
starting....
Entering initLog()...
Commands registered!
initLog() done!
Warn:CFG:CFG_InitAndLoad: Correct config has been loaded with 1 changes count.
Info:MAIN:Using SSID []
Info:MAIN:Using Pass []
Info:MQTT:MQTT_RegisterCallback called for bT w600CD1002B3/ subT w600CD1002B3/+/set
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/w600CD1002B3/ subT cmnd/w600CD1002B3/+
Info:GEN:PIN_SetupPins pins have been set up.
Info:MAIN:Time 1, idle 0/s, free 30488, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/8
Info:MAIN:Time 2, idle 0/s, free 30488, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/8
Info:MAIN:Time 3, idle 0/s, free 30488, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/8
Info:MAIN:Time 4, idle 0/s, free 30488, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/8
 ap create Successfully ! 

w600 Compiled @Nov 15 2022 14:45:49
delaying start
#Startup delayed 0ms#
#Startup delayed 10ms#
#Startup delayed 20ms#
#Startup delayed 30ms#
#Startup delayed 40ms#
#Startup delayed 50ms#
#Startup delayed 60ms#
#Startup delayed 70ms#
#Startup delayed 80ms#
#Startup delayed 90ms#
#Startup delayed 100ms#
#Startup delayed 110ms#
#Startup delayed 120ms#
#Startup delayed 130ms#`
And that repeats over and over. I tried refreshing it a few times even with different versions but the result is always the same.

openshwprojects, 2022-11-29:
Thank you for reporting. @iprak , can you check? I will also find my W600 to check.

Does it also really happen with earlier versions?

mikimause9, 2022-11-29:
Yes, I tried some older versions and I came as far as 1.14.108. The boot looping was present in all of them. The module restarts right after creating an access point.

mikimause9, 2022-11-29:
Sadly I have no other module to check if this is present as well. This may be caused by a broken module.

openshwprojects, 2022-11-29:
Can you please check more older versions? I will try to dig out my W600 tomorrow and check as well.

mikimause9, 2022-11-29:
Honestly, after checking the oldest version I could find for w600 (1.13.0) I am nearly sure that my module is broken. The behavior is identical to before.

openshwprojects, 2022-11-29:
maybe not enough current?

mikimause9, 2022-11-30:
I’ve checked on Booth laptop USB and 2A charging brick. It shouldn't be the case sadly.

iprak, 2022-11-30:
I have a couple switches running on 1.15.34 and they seem to be fine, they have been running for past 8 days now. Sorry I don't have a W600 with me which I can flash older firmware.

mikimause9, 2022-11-30:
I will just close this issue as this probably is my module's fault. Thanks for your help anyways!

closed 2022-11-30 by mikimause9

mikimause9, 2022-11-30:
Just to clarify I've put different software on this module and it didn't work as well. This was my module's fault :)

iprak, 2022-12-01:
OK. I found my module (W600 TW-02) and it has 1.15.36 on it. I updated it to 1.15.119. No issues.