openshwprojects/OpenBK7231T_App #509: OpenBK doesn’t reconnect after router reboots

ferbulous, 2022-11-30:
**Describe the bug**
I have a daily reboot schedule reboot for my wifi router at 5am. When I’m checking homeassistant, all the openbk devices are offline since 5am and I had to power cycle to get them connected again.
Is there something that I need to enable in the settings to force reconnect to wifi?

**Firmware:**
- Version [1.15.111]
- Device - Wall Switch/Light Bulbs
- Chip/model: BK7231N
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot wifi router
2. Openbk still doesn’t reconnect after router is online
3. Power cycle required to reconnect to wifi

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2022-11-30:
It should reconnect by default. @btsimonh / @valeklubomir , thoughts?
But I admit I had somehow a similar of issue in the past with my old crappy router.
The solution was to create "ping watchdog".
[image]

It was a long time since I needed that, can you check if it works?

openshwprojects, 2022-11-30:
@ferbulous making timeout longer for ping watchdog will not make it reconnect faster. If anything, it will make it reconnect slower.

Are you sure that it takes about 20 minutes for that device to reconnect?

Wait, maybe you have the same issue as me a month ago and just your router can't keep up with so many devices?

I had a problem once where devices didn't want to connect to router and it turned out that my old TP link can't handle too many clients at once. I replaced my router with newer model and it helped.

Btw @ferbulous  how are device groups? You have that black light issue, right?

ferbulous, 2022-11-30:
> Are you sure that it takes about 20 minutes for that device to reconnect?

Based on the homeassistant history logs

> Wait, maybe you have the same issue as me a month ago and just your router can't keep up with so many devices?

Not sure if router related since I do have mesh wifi setup, I did earlier try flashing esphome (libretuya) and it reconnect just fine after router reboot.

> Btw @ferbulous how are device groups? You have that black light issue, right?

The black color/light is still there on latest update. I'll add more comments on #503

openshwprojects, 2022-11-30:
Is it T or N?

WiredLife, 2022-12-07:
Hey, i can confirm this bug, looks like a race condition.
I reboot my two Belkin RT3200 with newest OpenWRT on it, then minimum of 1 from 9 lightbulbs doesnt reconnect, so i need to powercycle it.
Sometimes 2 or more.
Maybe it has something to do with 802.11r (roaming)?
Like it can not chosse which channel to connect?
I dont know

openshwprojects, 2022-12-13:
That's a very nice script and I can confirm that it should indeed work. I will just note that you don't have to do "01" index, it's simply 1.

@btsimonh do you know everything about this issue? I am checking my devices but all are up...

btsimonh, 2022-12-14:
I don't reboot my router often.  But I have booted my N device from the router to cause reconnect on occasion.  
I have concerns about DHCP/DNS - same LWIP threading issue as MQTT.
Also, MQTT 'gethostbyname' stalls the timer thread for a few seconds - I have a mod on the way which polls for resolution rather than wait.
If may be a good test to specify the MQTT server by IP - this would avoid the 'gethostbyname' wait.
@ferbulous - if you get a chance to setup as IP based mqtt config and retry, that may be useful diagnosis.