openshwprojects/OpenBK7231T_App #510: Future request Tasmota and/or Domoticz MQTT support

GravityRZ, 2022-11-30:
probably a long shot.
currently OpenBK is only MQTT Home Assistant supported.

i am trying to get things to work under Domoticz but it is a real pain, especially for color light bulbs
domoticz has plugin support for shelly and tasmota, not for home assistant.
they say it is home assistant compatible but it is really not

is there any chance we will support Tasmota in the future, seems logical because OpenBK is an tasmota alternative of esp8266 platforms.

openshwprojects, 2022-11-30:
@GravityRZ  if Domoticz MQTT protocol is not too complicated, I could do it very quickly, but I would need a real samples of what is supposed to be sent and received by OBK in order to work with Domoticz.

Can you provide me that? 

As far as I read, domoticz is using a single topic and expects a json data bunch with some kind of id and variables.

GravityRZ, 2022-11-30:
with domoticz you can use MQTT device in several ways.
1 by using a plugin which recognises MQTT devices
there are plugins for Shelly, Tasmota and several other solutions.

2 by using the MQTT Auto discovery plugin which creates devices when data is received
it supports both zigbee2MQTT zwaveJs2MQTT, Home assistant and tasmota.
The problem is it is not 100% correct with home assistant and tasmota

when MQTT info is received the device is ceated automatically.
the problem is i can not find anything on how this works or what i needs.
i did some testing with home assistant compatible code but i noticed that this does not work 100%

i am able to send a string which creates let's say a lightbulb dimmer but even that one does not work 100%
if i try to create a color bulb it does not work at all.
the problem is that with home assistant you are able to put things in a yaml config file and HA uses rules on the variables. as far as i can tell domoticz does not accept formulas like 

```
    rgb_command_template: "{{ '#%02x%02x%02x0000' | format(red, green, blue)}}"
    rgb_value_template: "{{ value[0:2]|int(base=16) }},{{ value[2:4]|int(base=16) }},{{ value[4:6]|int(base=16) }}"
```

if i publish the code below in  the mentioned topic domoticz will create a dimmable lightbulb but percentages are not correct or do not stick
looking at the example it looks like it is not that difficult but if we do not know what makes MQTT AD in domoticz tick we really do not know what it needs to create a certain device

Topic: homeassistant/light/OpenBK7231T_17A1C483_light/config

```
      {
        "unique_id": "OpenBK7231T_17A1C483_light",
        "name": "lsclamp",
        "command_topic": "cmnd/lsclamp/led_enableAll",
        "availability_topic": "lsclamp/connected",
        "state_topic": "lsclamp/led_enableAll/get",
        "availability_topic": "lsclamp/connected",
        "payload_on": 1,
        "payload_off": 0,
        "brightness_command_topic": "cmnd/lsclamp/led_dimmer",
        "brightness_value_template": "{{ value }}",
        "brightness_scale": 100,
        "brightness_state_topic": "lsclamp/led_dimmer/get",
        "on_command_type": "brightness",
        "state_value_template": "{{ value }}"
      }
```

GravityRZ, 2022-12-01:
i am asking the domoticz community to see if i can get some help
basically what we need is a domoticz MQTT AD page like the one from HA
https://www.home-assistant.io/integrations/light.mqtt

i however tried some simple things and they do not work 100%.

this seems to be an interesting script to see the config and command topics HA is expecting for lights [https://github.com/home-assistant/core/ ... t_light.py](https://github.com/home-assistant/core/blob/dev/tests/components/mqtt/test_light.py)

I think Domoticz is expecting something like "test_light_rgb/rgb/set", "255,255,255"
And your device is sending probably something like test_light_rgb/rgb/set', '#ffffff'

if OpenBK is following the HA standard why are  the state topic and the command topic not in the same group/subgroup

eg with HA and Tasmota

command_topic: cmnd/lsclamp/led_dimmer/
state_topic: lsclamp/led_dimmer/get

should this not be more logical
command_topic: lsclamp/led_dimmer/
state_topic: lsclamp/led_dimmer/get

this is not the cause of the problem but if the path is the same then the ~ prefix can be used which makes the commands shorter and easier to read.

GravityRZ, 2022-12-05:
@openshwprojects 
can you check what kind of homeassistant implementation is used?
i studied the code on this page [https://github.com/home-assistant/core/blob/dev/tests/components/mqtt/test_light.py](url)
it looks HA needs separate values for r,g,b,w,w
i know that domoticz also likes this
is it an idea to supply these numbers seperate or in one string seperated by comma

openshwprojects, 2023-01-29:
@GravityRZ we added ioBroker support, can you help testing, we also now do TELE publishes (if enabled in option), is this what you need?

GravityRZ, 2023-01-30:
do you mean it is tasmota compatible?

if so i can install the tasmota plugin and see how the light is recognised

vinibali, 2024-02-23:
My feature request for somewhat the same topic:
https://github.com/openshwprojects/OpenBK7231T_App/issues/1100

vinibali, 2024-09-29:
@GravityRZ have you tried that so far?