openshwprojects/OpenBK7231T_App #511: Can not connect to wi-fi network without password

v1ct0r2005, 2022-12-02:
Can not connect to wi-fi network without password
(with blank password)

- Version 1.15.119
- Device Aubess Mini smart switch 16A
- Chip/model: BK7231N

Steps to reproduce the behavior:
1. Clear password field on your router
2. Enter only SSID on your divice config, leave blank password field
3. Save and reboot your device

After reboot device start in AP mode

openshwprojects, 2022-12-02:
You are correct. There was indeed such a behaviour implemented.
I have fixed it now, now only  empty SSID should trigger AP mode.
Can you check latest version?
https://github.com/openshwprojects/OpenBK7231T_App/commit/6dc7510f77fcf206767e5674d47db1f2845c819b

v1ct0r2005, 2022-12-02:
In version 1.15.124 works great!
Thank you very much for fixing it!

closed 2022-12-02 by v1ct0r2005