openshwprojects/OpenBK7231T_App #514: Feature Request / Howto / MQTT for iobroker Sonoff Plugin

christi-ko, 2022-12-04:
The first OpenBK7231N Powerplug found the way into my iobroker Smart Home. After being able to disconnect it from the net by using cloudcutter and now flashing it with OpenBeken i am stuck with getting it to work with my iobroker sonoff plugin:

My Hardware:
Smart Plug (Aubess) | BK7231N (CB2S) | switch,plug,power monitoring,BL093

The Plugin i am using in iobroker is the following:
[https://github.com/ioBroker/ioBroker.sonoff](https://github.com/ioBroker/ioBroker.sonoff)

However i am not able to switch to another plugin right now, but i would like to prefer to have OpenBeken to act similar like a standard Tasmota device would do ...

For now i get several failure messages:
`sonoff.0 | 2022-12-04 19:17:46.362 | warn | Client [BK7231N_1] Message 439 deleted after 11 retries
-- | -- | -- | --
sonoff.0 | 2022-12-04 19:17:46.361 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/energycounter/get, payload: 25.634321
sonoff.0 | 2022-12-04 19:17:46.361 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/energycounter_last_hour/get, payload: 0.000000
sonoff.0 | 2022-12-04 19:17:46.361 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/voltage/get, payload: 226.035308
javascript.0 | 2022-12-04 19:17:46.140 | debug | script.js.common.Strom.Fernseher_Auto_aus: Ich bin innerhalb der Zeit, bleib wie ich bin
sonoff.0 | 2022-12-04 19:17:44.360 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/current/get, payload: 0.007794
sonoff.0 | 2022-12-04 19:17:44.360 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/energycounter/get, payload: 25.634321
sonoff.0 | 2022-12-04 19:17:44.359 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/energycounter_last_hour/get, payload: 0.000000
sonoff.0 | 2022-12-04 19:17:43.664 | debug | [BK7231N_1] Received: DVES_BK7231_N1/voltage/get = 226.035308
sonoff.0 | 2022-12-04 19:17:42.359 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/current/get, payload: 0.007794
sonoff.0 | 2022-12-04 19:17:42.359 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/energycounter/get, payload: 25.634321
sonoff.0 | 2022-12-04 19:17:42.359 | debug | Client [BK7231N_1] Resend message topic: DVES_BK7231_N1/energycounter_last_hour/get, payload: 0.000000
sonoff.0 | 2022-12-04 19:17:40.829 | debug | Client [DVES_678555] pingreq
`

Has anyone of you being able to get it work? with iobroker and sonoff plugin. I guess the issue is the topic and topic naming which is different compared to tasmota, but i am not sure ....

openshwprojects, 2022-12-29:
I am looking into it, can you specificy which exact Tasmota MQTT messages you need?

christi-ko, 2022-12-29:
> I am looking into it, can you specificy which exact Tasmota MQTT messages you need?

Thank you for looking into it.
I think it's the way the Topic is named. The plugin in expecting the topic name to be in a certain manner. I would propose the following according above mentioned Homepage [https://github.com/ioBroker/ioBroker.sonoff](https://github.com/ioBroker/ioBroker.sonoff):
`tele/DeviceNAME/ENERGY`

openshwprojects, 2022-12-29:
I saw the list you linked, but it's very, very large:
[image]
There are even things like:
```
/DeviceNAME/BM280/RestartReason
/DeviceNAME/BM280/CarbonDioxide
```
my question is, which MQTT topics from that list you need for your device. Ok, you need tele/DeviceName/ENERGY, what else is needed for you?

I think I can get them up and running for you relatively quickly, maybe even few days, but I need specific details what you need.

christi-ko, 2022-12-29:
This all are examples, which messages can be handeled. I want to use it with a PowerPlug. Therefore the one used for Power would be the best i guess. (But not important, just pick one of them):
`tele/DeviceNAME/POWERx`

openshwprojects, 2022-12-29:
POWER is to set relay on and off.
ENERGY is to get data from sensor (I think so),
so you need those two?

Would you be able to provide a sample of what is returned by those topics? But don't worry if you can't, I will find my BW-SHP6...

christi-ko, 2022-12-30:
Hi, this is also matching to expected format. I also did it that "easy" way for my "little" weatherstation project in Arduino.  Exactly this format would be good ...

Hahnweiler, 2023-01-12:
"POWER is to set relay on and off."
Is this already implemented? I flashed a LCS Smart Power Plug and can see the Objekt "_id": "sonoff.0.obkBF31F28A.**alive**", at the Sonoff Plugin (IOBROKER). But i miss the objekt "POWER" to switch the powerplug on and off.

openshwprojects, 2023-01-24:
@Hahnweiler I think that POWER works, since yesterday it also returns on stat the current power on MQTT.
What else is needed?

I would like to support iobroker plugin.

Can someone help with testing?

openshwprojects, 2023-01-24:
@Hahnweiler  @christi-ko  it seems it's working, would you like to help testing?

openshwprojects, 2023-01-24:
https://github.com/openshwprojects/OpenBK7231T_App/issues/634

closed 2023-01-24 by openshwprojects

christi-ko, 2023-01-24:
Hi, just downloaded the newest version ready for testing. But now i saw the flag is still missing, right?