openshwprojects/OpenBK7231T_App #515: channelSet fails if topic has '

no2chem, 2022-12-04:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version [e.g. 1.14.129]
- Device? 
- Chip/model: BK7231T?
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2022-12-04:
I have never tried setting such device name. There indeed might be an issue.

Would you like me to help, or should I wait for your PR?

no2chem, 2022-12-04:
I'm trying to setup the build toolchain, admittedly its taking me awhile. I can send the PR, it's just a minor fix but I haven't tested it. I'll do that now.

openshwprojects, 2022-12-06:
@no2chem  was it tested on your device? I will try to check it in practise today, but it looks ok

no2chem, 2022-12-06:
i'm testing it at the moment, there's a bug so standby :)

no2chem, 2022-12-06:
@openshwprojects fixed, please take a look at latest push

PR #517 (2022-12-08): fix: setChannel should parse the client
If the mqtt client contains a '/', setChannel will fail to find the channel because it will stop at the first '/'.

This PR fixes setChannel to scan the client name instead, so it correctly skips over client. 

A new function `MQTT_RemoveClientFromTopic` is added, which skips over the topic.
In addition, the parsing in `channelSet` is greatly simplified, and and unnecessary copies are removed.

Fixes #515
closed 2022-12-08 by openshwprojects