openshwprojects/OpenBK7231T_App #516: please add hsb

WiredLife, 2022-12-04:
the conversion from hsb to rgb is not flawless
tasmota has support for hsb, why openbk not?
please add ='(

openshwprojects, 2022-12-05:
Hello, can you give a specific example where the HSB conversion goes wrong, both with "wrong" output and the expected one?

WiredLife, 2022-12-06:
i mean, there is no support to send a hsb value to openbk, only rgb, which needs a conversion before send to a lamp

openshwprojects, 2022-12-06:
Oh ok, I will add this for you today, but can you specify some things?

First of all, does sending HSBColor to light that is OFF in Tasmota, enables this light? 

If light is in temperature mode, does sending HSBColor switches it back to RGB mode?

WiredLife, 2022-12-06:
> First of all, does sending HSBColor to light that is OFF in Tasmota, enables this light?
No only sends the color
> If light is in temperature mode, does sending HSBColor switches it back to RGB mode?
Yes it switch from temperature mode to RGB mode

You're great, thanks!

openshwprojects, 2022-12-07:
Added HSBColor 360,100,100 (syntax), also HSBColor1 2 and 3.
**Can you check if it works like in Tasmota?**
Thanks

WiredLife, 2022-12-07:
it works fine but only over the console
is mqtt not added yet?

openshwprojects, 2022-12-07:
@WiredLife  what is the MQTT syntax for that? 

This is an OpenBeken command, it should work if you send it as an OpenBeken command through MQTT. But maybe you expect different syntax. I can add that, if I know the syntax.

Our syntax example:
    rgb_command_topic: "cmnd/obk760BF030/led_basecolor_rgb"
In case of HSB: 
"cmnd/obk760BF030/HSBColor" with a proper payload..

WiredLife, 2022-12-07:
i send a string with 360,100,100 to cmnd/Spiegelschrank_Lampe1/HSBColor but it does nothing
also nothing on the console log

openshwprojects, 2022-12-07:
@WiredLife what happens if you send:
- HSBColor with value 360 100 100 (without ,)
- HSBColor1 with value 360 (or 90) ?

openshwprojects, 2022-12-07:
ah ok, no problem, is there anything else useful we could add?

WiredLife, 2022-12-07:
i will dig deeper into this nice stuff
just bought many different smart lightbulbs and stuff, dump the firmware, use tuyacloudcutter to create an exploit and then flash youre great software
i will come back if i miss a feature
currently its working nice and sweet without that garbage china cloud thingi and my openhab system

WiredLife, 2022-12-07:
ah nice, sure i will do, this project is great!
i'm from germany
will add some infos the next days in my free time
i've written down some gpio pinouts and stuff from the lamps i got working

WiredLife, 2022-12-07:
do you know if its possible to implement wifi 802.11w (protected management frames)?
i dont know if tasmota has it already or its not possible because less storage

openshwprojects, 2022-12-07:
Doing such thing from scratch would take a lot of time, we would have to check if Beken SDK supports it.

Oh indeed, that call was missing, I added it now, for all 4 commands (HSBColor, HSBColor1 and 2 and 3), can you recheck?

WiredLife, 2022-12-07:
yep now its working fine
thanks!

openshwprojects, 2022-12-07:
@GravityRZ HSBColor is not send, this is just not implemented.

If you need it, please tell me, and I will implement it for you, but I need to know the exact syntax you'd like to get.

GravityRZ, 2022-12-07:
ok. firsdt let me see if i can work with HSBColor to see if my light will support it

WiredLife, 2022-12-08:
I think this issue is not a hsb problem but maybe fit in here too
On the original firmware with tuya-mqtt i got much more better colors, they look exact like in the color picker circle of openhab
With openbk only the outside colors fit
I already tested higher brightness but no change
---Original---
[image]
---openbk---
[image]
---color circle---
[image]

WiredLife, 2022-12-08:
maybe this https://github.com/openshwprojects/OpenBK7231T_App/pull/447 will fix this?
can anyone provide a OpenBK7231T ota image with this?

HumbleDeer, 2022-12-09:
> First of all, does sending HSBColor to light that is OFF in Tasmota, enables this light?

Tasmota has features like OpenBK to enable special features where "turn on on color change" is one

HumbleDeer, 2022-12-09:
> I think this issue is not a hsb problem but maybe fit in here too On the original firmware with tuya-mqtt i got much more better colors, they look exact like in the color picker circle of openhab With openbk only the outside colors fit I already tested higher brightness but no change 

I can second this with my own experience yesterday. Colours don't look quite right on all levels (H, S, and L) when sent from Hass

openshwprojects, 2023-01-12:
Please continue discussion here: https://github.com/openshwprojects/OpenBK7231T_App/issues/599

closed 2023-01-12 by openshwprojects