openshwprojects/OpenBK7231T_App #518: W801 Micro HLK-W801-Kit for 2EUR.

valeklubomir, 2022-12-06:
I today received HK-W801-Kit
[image]
with chipset
[image]
Managed to build and flash with OpenW800 and OpenBK7231T_App.
[image]

Findings:
- None of the src/drivers are included. 
- LWIP 2.0.3 version
- No I2C
- No All pina are listed in config module

All-in-All good start

openshwprojects, 2022-12-07:
Very nice board. I also have one W800, but my one is smaller and more crude.

Regarding drivers:
- they are disabled because I have never heard about any W800 device needing a driver
- the plan is to support #ifdef enable/disable for each driver to make configuration easier (obk_config.h ? )

I am not sure about missing pins. It's worth investigating.

It might be also worth to do a stability check on W800. I didn't have time to do it yet, but from what I remember, the W800 stability is not bad.
BL602 had a stability issue, but I fixed it thanks to @valeklubomir idea:  https://github.com/openshwprojects/OpenBL602/commit/c73dcaab7e73ab68b29bd20a3b5d4c0992902c1d
(above commit, as crude as it is, made it totally stable according to some user reports)

W800 and W600 both supports OTA, but only from HTTP page - by putting a link to HTTP server with binary file.
It might be worth looking into adding a support for Web App OTA to those platforms.

If you have some time, you can also post this device to our Elektroda forum - so I can add it to the list: https://openbekeniot.github.io/webapp/devicesList.html . If you are from EU, you might be also able to receive a free gift of Elektroda gadgets - SD card and stickers. Just message @gulson  there after posting an article/teardown/review.

**In general, any help is welcome.**  I am currently preparing the Simulator for the release (with a crude schematic drawing and interactions) so both people and devs can easily test drivers, design scripts, etc on PC (currently on Windows).

openshwprojects, 2022-12-07:
Really, one can buy CB2S now on ali? Can you post a link? I will check, but it sounds like a good news.

I already tried to buy BK7231 on ali once and I had a bad experience.
[image]
I ordered BK7231U module and got that:
[image]

valeklubomir, 2022-12-07:
Sorry small mistake it is CB3S [https://www.aliexpress.com/item/1005004788201798.html](https://www.aliexpress.com/item/1005004788201798.html)
But I just ordered it, waiting for delivery, and I hope I will not experience same situation like you, 
and when it is written CB3S,  it will be CB3S inside. 
There is always some risk with ali. But from my experience 99% purchases where correct.

closed 2022-12-09 by valeklubomir

openshwprojects, 2022-12-09:
Btw @valeklubomir , before I forget, I saw that you still have some nice changes in the TuyaMCU code etc etc.

Can you please prepare a Pull Request soon? I am doing a lot of changes these days and I don't really want our forks to drift away from each other too far.

btsimonh, 2022-12-09:
@valeklubomir - if looking at Tuya, pls be aware of the Tasmota changes recently merged.
I re-worked Tas to have a valid state machine for the tuya protocol..... it fixed issues with crashing some Tuya MCU - the problem being that once they had crashed, the only way out was a power-off, and that's not convenient when it's a wall switch.