openshwprojects/OpenBK7231T_App #520: MQTT not started when executing "Start Home Assistant Discovery" with Firmware 1.15.149

NielsPiersma, 2022-12-07:
After flashing 1.15.149 Start Home Assistant Discovery stopped working and the devices disappear from Home Assistant.

**Firmware:**
- Version [1.15.149]
- Device: WOOX R6080 Smart Plug
- Chip/model: BK7231T
- Device config: 
    - 6 LED_n 1
    - 10 Btn 1
    - Rel 1
-

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 1.15.149
2. Click on 'Start Home Assistant Discovery'
3. See message "MQTT not Running"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[image]

Downgraded back to 1.15.120, working again.
[image]

Could not really find where / how this is a configuration setting.
Niels

openshwprojects, 2022-12-07:
Hello, can you narrow down a bit which release started having the problem, for instance, is 1.15.135 working?

NielsPiersma, 2022-12-07:
Yes, will try a few flashes 135 is working. 148 is not.

Will report asap

openshwprojects, 2022-12-07:
while reporting, please also show the mqtt status from the main page.... If it's connected or not

NielsPiersma, 2022-12-08:
@openshwprojects,

We just flashed 3 units to Firmware 1.15.157

<html>
<body>
<div id="state" style="padding: 0px; font-size: 16px; margin: 0px 0px 0.2em; color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(33, 51, 62); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><h5>1 drivers active (DGR), total 17</h5><h5>Channel 1 = 0</h5><h5>Cfg size: 2016, change counter: 21, ota counter: 0, boot incompletes 0 (might change to 0 if you wait to 30 sec)!</h5><h5>Ping watchdog - 0 lost, 0 ok!</h5><h5>Wifi RSSI: Excellent (-47dBm)</h5><h5>MQTT State: connected RES: 0(ERR_OK)<br>MQTT ErrMsg:<br>MQTT Stats:CONN: 1 PUB: 4 RECV: 0 ERR: 0</h5></div><br class="Apple-interchange-newline">
</body>
</html>

Build on Dec 8 2022 17:28:05 version 1.15.157
Online for 15 minutes and 15 seconds
Device MAC: 7C:F6:66:49:83:80
Short name: woox-01, Chipset BK7231T

I would say, issue resolved.
Niels

closed 2022-12-08 by NielsPiersma