openshwprojects/OpenBK7231T_App #521: BL602 (Sonoff S40) unable to connect to WiFi

ArcadeMachinist, 2022-12-08:
**Describe the bug**
Module is unable to connect to WiFi.
Initial module provided hostap works, I can reach WebUI and input WiFi configuration.
Module never connects.
My WiFi is ruuning on channels 11 and 6.
Below I'm attaching BL602 debug log, from what I understand, it does say it can't find a valid BSSID to connect to, because scan fails. 

**Firmware:**
- Version OpenBL602_1.15.149 and OpenBL602_1.15.140
- Device Sonoff S40 with power stats
- Chip/model: BL602
- Device config None

**To Reproduce**
Steps to reproduce the behavior:
1. Go to amazon.com, buy Sonoff S40 WiFi
2. Open up, solder wires, flash OpenBL602_1.15.149
3. Connect to HostAP, configure WiFi
4. Check device log for WiFi connection loop

[BL602-Log.txt](https://github.com/openshwprojects/OpenBK7231T_App/files/10180992/BL602-Log.txt)

openshwprojects, 2022-12-08:
Hello, can you check some previous versions in order to let us know when the issue started to happen? I don't recall making any serious changes to BL602 handling, but one never knows....

ArcadeMachinist, 2022-12-11:
Tried to go back to OpenBL602_1.15.130 and then all the way to 1.15.20
It can't find any BSSID to join.
Tried different networks and even setting up an old standalone router.
Tried erasing the chip completly, etc.

May be some new chip revision?
Also should MAC be 00:00:00:00:00:00 on AP scan?
I don't have other BL602 based devices for reference.

[image]

```
[22:44:01.900] - [WF][SM] State Action ###disconnect### --->>> ###connecting###
[22:44:01.900] - [WF][SM]  Action Connect
[22:44:01.900] -            ssid AIOT3930
[22:44:01.900] -            ssid len 8
[22:44:01.900] -            psk **********
[22:44:01.900] -            psk len 10
[22:44:01.900] -            pmk
[22:44:01.900] -            pmk len 0
[22:44:01.915] -            mac 00:00:00:00:00:00
[22:44:01.915] - connecting using vif_idx 0
[22:44:01.915] - ===start sm_get_bss_params===
[22:44:01.915] - bssid[0] = 0xffff
[22:44:01.915] - bssid[1] = 0xffff
[22:44:01.915] - bssid[2] = 0xffff
[22:44:01.915] - search ssid = AIOT3930
[22:44:01.915] - result ssid index = -1
[22:44:01.915] - ===en  sm_get_bss_params===
[22:44:01.930] - [WF][SM] Entering connecting state
[22:44:01.930] - [APP] [EVT] Reconnect 88357
[22:44:01.930] - [APP] [EVT] Reconnect 58357
[22:44:01.930] - [APP] [EVT] Connecting 58357
[22:44:01.930] - [APP] [EVT] Connecting 58357
[22:44:01.930] - Info:MAIN:Main_OnWiFiStatusChange - WIFI_SAA_CONNECTING
[22:44:02.291] - Info:MAIN:Time 58, idle 0/s, free 106168, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21
[22:44:03.290] - Info:MAIN:Time 59, idle 0/s, free 106168, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21
[22:44:04.290] - Info:MAIN:Time 60, idle 0/s, free 106168, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21
[22:44:05.134] - ===start sm_get_bss_params===
[22:44:05.149] - bssid[0] = 0xffff
[22:44:05.149] - bssid[1] = 0xffff
[22:44:05.149] - bssid[2] = 0xffff
[22:44:05.149] - search ssid = AIOT3930
[22:44:05.149] - result ssid index = -1
[22:44:05.149] - ===end sm_get_bss_params===
[22:44:05.165] - ind ix 0x42029524, chan_ctxt is 0x00000000
[22:44:05.165] - connect failure, ssid = AIOT3930, index = -1
[22:44:05.165] - [RX] Connection Status
[22:44:05.165] - [RX]   status_code 12
[22:44:05.165] - [RX]   connect result: SSID error, scan no bssid and channel
[22:44:05.165] - [RX]   MAC 00:00:00:00:00:00
[22:44:05.165] - [RX]   vif_idx 0
[22:44:05.165] - [RX]   ap_idx 255
[22:44:05.165] - [RX]   ch_idx 0
[22:44:05.181] - [RX]   qos 0
[22:44:05.181] - [RX]   acm 0
[22:44:05.181] - [RX]   assoc_req_ie_len 0
[22:44:05.181] - [RX]   assoc_rsp_ie_len 0
[22:44:05.181] - [RX]   aid 0
[22:44:05.181] - [RX]   band 0
[22:44:05.181] - [RX]   center_freq 0
[22:44:05.181] - [RX]   width 0
[22:44:05.196] - [RX]   center_freq1 0
[22:44:05.196] - [RX]   center_freq2 0
[22:44:05.196] - [WF][SM] Exiting connecting state
[22:44:05.196] - [WF][SM] State Action ###connecting### --->>> ###disconnect###
[22:44:05.196] - [WF]SSM] Entering disconnect state
[22:44:05.196] - [WF][SM] Will retry in 2000 ticks
[22:44:05.196] - [APP] [EVT] disconnect 61620, Reason: SSID error, scan no bssid and channel
```

ArcadeMachinist, 2022-12-16:
I can mail the device in question to anyone interested, got 4-pack of the same.