openshwprojects/OpenBK7231T_App #523: Home Assistant auto-discovery, like ESPHome?

TurkeyMan, 2022-12-08:
Just wondering, does OpenBeken have the Home Assistant auto-discovery using the HA protocol like ESPHome does? It's much simpler than MQTT.

davygravy, 2022-12-10:
This is already present. 

In the webif of your flashed smart plug/device:  Config > Home Assistant Configuration > Start Home Assistant Discovery

After these button presses, my devices appeared automatically in my MQTT integration in HA. I didn’t have to touch my HA configuration.yaml file, and the whole process was essentially effortless.

euro2, 2022-12-13:
does this give any benefits in regards to visibility in HA? When i reboot HA i have to manually toggle all openbeken's once to get HA to know the status of them

openshwprojects, 2022-12-27:
@euro2 enable "broadcast self state every minute" or try something like retain?

openshwprojects, 2023-01-08:
@iprak  is doing great things with autodiscovery, this is getting very advanced

closed 2023-01-08 by openshwprojects

rojoricardo, 2025-04-13:
Hi, my autodiscovery for home assistant is not working, any ideas? I can see my device in MQTT Explorer, but home assistant is not auto discovering it... I really dont know whats going on