openshwprojects/OpenBK7231T_App #524: Revert to factory firmware feature

TurkeyMan, 2022-12-08:
This is an aspirational goal, but it has come up for me twice now, and probably more than more gadgets I have...

I have a lot of gadgets, and some of them I don't use. I've had other people ask me about them, and I would give them, but these other people probably don't want to run DIY automation. They actually want the device returned to its stock firmware to use with whatever phone app and web service it prefers.
Now, I don't make an effort to store off each devices stock firmware... but it made me think maybe the flash/install script could/should do this automatically.

Consider:
For devices where the OpenBK image is smaller than the device flash memory; do we have memory to spare?
Before flashing, read the stock firmware image. Compress the binary with strong compression (I just compressed stock image with 7z and it was 40% original side). Then, when flashing the new firmware, also write the compressed stock firmware to the flash after the new image, and write the offset of the factory reset image out somewhere.
In OpenBK, add a "Factory Reset" button, which would find the image at the offset that was recorded, decompress it and write it back to the flash memory.

This would be a real pro-move, and definitely make it feel like a polished product. The only limitation is if OpenBK is small enough to also cram a compressed stock firmware image into the flash?

For the device I just flashed; the stock firmware I extracted was a 2mb dump. I compressed it down to ~900kb. I noticed that OpenBK7231N_QIO_1.15.152.bin is 1,192kb. Assuming it's a 2mb flash (judging from the stock firmware dump size), then this is really close to fitting. Perhaps OpenBK can be optimised a small amount and it would fit...?

Anyway, just a thought. I think it'd be good to have an official revert function, so future owners of the device can restore it without to know about the processor, find firmware images, having to know how to disassemble and flash the device, etc...

btsimonh, 2022-12-08:
the issue with keeping the original in flash is that that spare space is the OTA area!! :(.
It may be possible to generate an OTA image from the original (using the tools in the SDK).  The question will then be what other 'stuff' do we need to restore?

openshwprojects, 2022-12-08:
I'd say that the general idea is good, but the complexity penalty is too large for that to be implemented in a foreseeable future. We have much more pending issues to do.

TurkeyMan, 2022-12-08:
Yes, I did say "aspirational"... I wouldn't imagine it's high priority, but definitely something to keep on the radar, and something to keep in mind; possibly it may affect other design decisions that could conflict with a goal like this...

btsimonh, 2022-12-08:
I like the idea of a formal backup process - @TurkeyMan maybe take a look at the Python and see what you can contribute? (I hate python.... the production of the flashing thing was a 'needs must' process!).

TurkeyMan, 2022-12-09:
Python is literally the most upsetting programming language that exists. Narrowly beating javascript as a close runner-up :P
I would be forced to rewrite the whole thing on principle :/

btsimonh, 2022-12-09:
If you could re-write in nodejs you would be my friend for life!!!

TurkeyMan, 2022-12-13:
I quite like vibe.d: https://vibed.org
I just don't see why, because web things, that they should be written in javascript like a webpage. Web services have no reason not to be real programs :P

btsimonh, 2022-12-13:
ok, 'D' is a new one on me!!! is it c+++? or a JS derivative?  I love JS because I can abuse it :).

TurkeyMan, 2022-12-13:
It's kinda like C++ if C++ weren't the worst ever and learned some lessons from the last 30 years. If you think JS can suffer your abuse, you know nothing of D ;)