openshwprojects/OpenBK7231T_App #527: Disabling REST api in safe mode makes restoring RF config impossible

rianadon, 2022-12-09:
It looks like in safe mode the webapp and associated api commands are disabled if the chip starts up in safe mode (and is thus in AP mode).

The mac address on my BK7231T board got corrupted with zeros after an OTA update, and I was hoping to use the webapp to reset the RF config. However, since the webapp is disabled in safe mode, I'm not sure how I can do this.

I think it would make sense to keep the webapp enabled in safe mode, then include instructions on how to run the webapp locally in the case that you are connected to the AP so the github.io address of the webapp does not load. For example, I was able to download the https://github.com/openbekenIOT/webapp repository, download Vue.js locally, then run it from an HTTP server.

openshwprojects, 2022-12-09:
@rianadon 
You are correct. That is also my concern and I was going to address it.

My solution for that was slightly different. 

I's suggest to add a command to enable rest API by hand, so even in safe mode people get manually get that running.

What do you think?

I can get you it working within a hour or so, just depending if you think it's a good solution?

rianadon, 2022-12-09:
I think your solution makes sense as well! I don't think it matters whether the rest API is enabled at startup or via a command inside safe mode—just as long as there is a way to use the web app in safe mode, since so much functionality is built into it. I'm actually starting to prefer using the web app over the built-in app for configuration. It's very convenient!

No rush, but props to you if you can create it that quickly!

rianadon, 2022-12-09:
Yes I can test the commit tomorrow! I think exiting safe mode would do the trick too... assuming it's safe to exit safe mode 😄

openshwprojects, 2022-12-09:
@rianadon thats a good point.
I have added a mechanism for that - the 'exit safe mode' will NOT run autoexec etc etc, so autoexec crash is out of the question.
Pins crash, on the other hand, can be fixed without exiting safe mode... hmmm...
I am thinking is there any other fault scenario with a relatively high probability of happening, but no idea yet... automatic driver start will also not happen when pressing 'exit safe mode'

rianadon, 2022-12-10:
I just had the chance to test version 1.15.160! I'm not too sure what's going on, but it looks like exiting safe mode is not successful, and the board still boots back into safe mode for me.

I'm viewing the logs with `nc <ip address> 9000` and receive these messages immediately after exiting safe mode:
```
Info:MAIN:[SAFE] Time 137, idle 535921/s, free 88464, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 6/38
Info:MAIN:Going to call Main_ForceUnsafeInit
Info:GEN:PIN_SetupPins pins have been set up.
Info:MQTT:MQTT_RegisterCallback called for bT light/ subT light/+/set
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/light/ subT cmnd/light/
```

After which the wifi AP kicks me off and the light blinks once (as opposed to the 4 times it does when booting up for the first time).

When I reconnect to the wifi and run `nc` again, I see the `[SAFE]` in brackets, so it looks like the device enters safe mode again.

openshwprojects, 2022-12-10:
Huh... it works for me. Maybe try with "logdelay 100" typed in HTTP command field, assuming that works in safe mode....

I am not sure, I might add an ability for you to enable step by step things and check what is failing.

It works for me on T... [v](https://www.youtube.com/watch?v=Vyqf0cJungo)

it is not resetting the "failed boots" counter, so you have a point, but its certainly not crashing for me as well...

@btsimonh , can you check on your T?

rianadon, 2022-12-10:
I'm having trouble connecting to the chip over UART. It's been a while since I flashed it, so there's a chance I'm using the wrong pins.

It does look like the failed boots counter is resetting. I see this after exiting safe mode: "You are in safe mode (AP mode) because full reboot failed 0 times.Pins, relays, etc are disabled.".

openshwprojects, 2022-12-10:
@rianadon something is crashing in your device and the question is what and why. The worse thing is that it works for me.
I will try to add tomorrow some mechanism to allow you start really only the REST interface and nothing else and we will try again...

rianadon, 2022-12-10:
I figured out what's crashing it. I had flag 18 for the LED transitions (the experimental one ... I should've known). Turning that option off allowed me to boot out of safe mode.

So there is a way to exit safe mode after all! The problem was not even related to the RF config (though I did run the restore RF config afterwards for a good measure (my RF config had been written to all `0xff` before, maybe from the update). So even if there was a way to start only the REST interface, that still would not have fixed the boot problem.

Thank you for your help! I'll look into why flag 18 is not working.

closed 2022-12-10 by rianadon

openshwprojects, 2022-12-10:
what kind of LED do you have? 5 PWMs or I2C driver?

rianadon, 2022-12-10:
It's PWM. I suspect #532 might have something to do with why startup fails. With only 2 pins configured and flag 18 turned on, the chip can start up correctly.

However, with all 5 PWMs configured, the board turns the lights on very brightly on startup, which might be drawing too much current and possibly resetting the chip.