openshwprojects/OpenBK7231T_App #528: Higher brightness after update

WiredLife, 2022-12-09:
I updated my two LSC Filament Bulbs from openbk 1.12.something to newest
Now the lamps are much brighter on 1%, which is very annoying at our sleeping room
Was there some change between the versions?
I dont know the correct old version anymore because i did the update on both

openshwprojects, 2022-12-09:
@WiredLife  good spot, it was requested by one user to make minimum brightness 1 instead of 0....

So, should we return to previous version?

openshwprojects, 2022-12-09:
Or maybe tthat's not it... but I don't think it's also the PR you linked.
Can you check which version exactly breaks brightness?
Are they PWM-based?

WiredLife, 2022-12-09:
Uh, without a hint its not so good to flash every version between 1.12.x and now 😅
Yes its PWM based

WiredLife, 2022-12-09:
Hey, i found that there are binarys of PR 437 and the 0.1 version is the one which fixes the problem on my Filament Bulbs. Great!
https://github.com/rogere66/OpenBK7231T_App/releases/download/20221123/OpenBK7231T_dev_20221123-brt-0.1.rbl

openshwprojects, 2022-12-10:
Really? Then I can add this for you, altough just the brightness calculation, not the breaking change of channel value to float.
But.... it was never yet added, so how is it possible that you already had good dimmer in the past?

WiredLife, 2022-12-10:
I have no idea, as i said 1.12.something was fine with dimmer and the lamps were nice dark at 1%.
Maybe something PWM related?
I dont have tested the 447 firmware on my rgb lamps yet, because it has no hsb support.
Now i try to setup i386 docker and compile the current openbk with 447, maybe this works.

openshwprojects, 2022-12-10:
Full 447 is a bit problematic because of the floats. Maybe if we take just the exponential code.... Let us try that:
[image]
here is the RBL, how good does it look for you?

[exp-test.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/10200851/exp-test.zip)

WiredLife, 2022-12-10:
yes this works!

my hope was that 447 fixes the colors, but i cant test it without hsbcolor

WiredLife, 2022-12-11:
@openshwprojects i finally did it
i merged 447 with current branch and compiled it
and what can i say? it did not break something, just works, with good dimmer! and a littebit better colors (or its placebo, i dont know)
maybe there is something left to do to make the colors look better
can you take a look at the tasmota way? are they working with #rgb values also?
maybe they use a way with hsb without converting into rgb?

openshwprojects, 2022-12-12:
@WiredLife I would really need to know if my exp-test.zip also fixes brightness, because my exp-test.zip comes without some int to float changes and would be way better to integrate for now than full 447. Do you think that my exp-test.zip would be enough?

I think they still need to convert to RGB in the end.

WiredLife, 2022-12-12:
As i said right after your comment with the exp-test.zip
"Yes this works!"

openshwprojects, 2022-12-14:
We have added led_expoMode command

closed 2022-12-14 by openshwprojects