openshwprojects/OpenBK7231T_App #529: Energy Monitoring on Sonoff S40 (BL602) - Feature Request/Driver Enable

davygravy, 2022-12-09:
**Describe the bug**
Sonoff S40 has the capability to monitor energy use via the CSE7759B chip (similar to CSE7766).  Makes sense to have this enabled and usable.

**Firmware:**
- Version 1.15.158
- Device  Sonoff S40
- Chip/model: BL602
- Device config - not applicable here

**To Reproduce**
Steps to reproduce the behavior:
- already flashed my 2 of my S40s with OpenBL602 and they are working well.  Just no way to turn on Energy Monitoring.

**Screenshots**
- not applicable here

**Additional context**
From elektroda.com forum:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20228945#20228945

Certainly willing to test/assist.

openshwprojects, 2022-12-10:
I'll take a look into that in the following days, I will most likely need you to copy the debug log from BL602, but that's when I add the UART read...

openshwprojects, 2022-12-29:
@davygravy , I replied on forum:
https://www.elektroda.com/rtvforum/topic3889041-210.html
my question is, which serial port do you need for that device? current,y uart0 is working for me - pins 16 and 7