openshwprojects/OpenBK7231T_App #530: SK6812

Ugly-Wan, 2022-12-09:
I have successfully flashed this product:
https://www.amazon.com/Gevcc-Changing-Corner-Google-Assistant/dp/B09Y1NSVNQ?th=1
https://manuals.plus/kyinflose/dfqy-cs14-corner-floor-lamp-manual

I have the buttons and the IR reviver working. The microphones I have set to ADC via the ADC pin but I don't seem to see any info in the logs to reflect the data is being picked up - anyway the mic is less on an issue.

It seems that the LED's are based on this https://cdn-shop.adafruit.com/product-files/1138/SK6812+LED+datasheet+.pdf

There is no additional IC and they apparel to be driven directly from the BK7231N via P16.

Do you have any experience with a similar devices?

Any help would be excellent thank you.

### Photos

[image]

[image]

[image]

### Pin Map

Pin number | Symbol | I/O type | Mapping | Function
-- | -- | -- | -- | --
1 | P14 | I/O | Music BTN | Common GPIO, which can be reused as SPI_SCK (Correspond to Pin 11 of the IC)
2 | P16 | I/O | RGB | Common GPIO, which can be reused as SPI_MOSI (Correspond to Pin 12 of the IC)
3 | P20 | I/O | ??? | Common GPIO (Correspond to Pin 20 of the IC)
4 | P22 | I/O | Mode BTN | Common GPIO (Correspond to Pin 18 of the IC)
5 | ADC | I/O | Microphone - output of A63A - LM321MF/X | ADC, which corresponds to P23 on the internal IC (Correspond to Pin 17 of the IC)
6 | RX2 | I/O | N/C | UART_RX2, which corresponds to P1 on the internal IC. (Correspond to Pin 28 of the IC)
7 | TX2 | I/O | N/C | UART_TX2, which is used for outputting logs and corresponds to P0 of the internal IC (Correspond to Pin 29 of the IC)
8 | P8 | I/O | ??? | Support hardware PWM (Correspond to Pin 24 of the IC)
9 | P7 | I/O | ??? | Support hardware PWM (Correspond to Pin 23 of the IC)
10 | P6 | I/O | IR | Support hardware PWM (Correspond to Pin 22 of the IC)
11 | P26 | I/O | ??? | Support hardware PWM (Correspond to Pin 15 of the IC)
12 | P24 | I/O | ??? | Support hardware PWM (Correspond to Pin 16 of the IC)
13 | GND | P | GND | Power supply reference ground
14 | 3V3 | P | 3.3v - from Vout AMS1117-3.3 | Power supply 3V3
15 | TX1 | I/O | TTL | UART_TX1, which is used for transmitting user data and corresponds to Pin 27 of the IC. For the MCU solution, please refer to CBx Module.
16 | RX1 | I/O | TTL | UART_RX1, which is used for receiving user data and corresponds to Pin 26 of the IC. For the MCU solution, please refer to CBx Module.
17 | P28 | I/O | N/C | Common GPIO (Correspond to Pin 10 of the IC)
18 | CEN | I/O | Reset | Reset pin, low active (internally pulled high), compatible with other modules (Correspond to Pin 21 of the IC)
19 | P9 | I/O | PWR BTN | Common GPIO (Correspond to Pin 25 of the IC)
20 | P17 | I/O | N/C | Common GPIO, which can be reused as SPI_MISO (Correspond to Pin 14 of the IC)
21 | P15 | I/O | ??? | Common GPIO, which can be reused as SPI_CS (Correspond to Pin 13 of the IC)
Test point | CSN | I/O |   | Mode selection pin. If it is connected to the ground before being powered on, enter the firmware test mode. If it is not connected or connected to VCC before being powered on, enter the firmware application mode. Correspond to Pin 19 on the internal IC.

btsimonh, 2022-12-09:
I have T based devices with 'music' input.
Apparently these don't have hardware FFT (according to beken rep in tiawan).
I'm interested in looking at the mic input - but there are more important hings at the moment :(.
It's a TV backlight, and my LEDs are PWM driven,
br,
Simon

Ugly-Wan, 2022-12-09:
Good to know that the mic input doesn't work, I will ignore that that completely for now.

openshwprojects, 2022-12-10:
Related: https://github.com/openshwprojects/OpenBK7231T_App/issues/497
Once 497 is solved (ramfunc? any more progress @valeklubomir  ? I myself didn't look into it more yet), we could add a simple drivers for all that addressable lights relatively easily.