openshwprojects/OpenBK7231T_App #532: Default color when enabling "Smooth transitions for LED" (flag 18)

Bacto, 2022-12-10:
Hi,

> First I want to thank you for the awesome work on this project. It's a game changer to me and my home automation 🤩

The "Smooth transitions for LED" flag is awesome:
- It is very pleasant to change color
- When I turn on the light the fade-in effect is perfect
- The light turns on **instantly** (while fade-in)

There is only one problem: the default color on start is cold (RGBCW is set to #000000FF00).
And I haven't found a way to change it.

I use this startup command: `backlog ; BP5758D_Map 1 0 2 3 4 ; led_dimmer 100 ; led_basecolor_rgbcw #000000FFFF ; led_enableAll 1`.
On start the bulb fade-in to cold white and then the script is executed and bulb fades to cold+warm white.

Is it possible to have a way to define the default color?

Adrien

openshwprojects, 2022-12-10:
Hello, I will try to help.
When looking at code I can see that default start color seems to be all off:
_float led_rawLerpCurrent[5] = { 0 };_
which doesn't seem to explain the behaviour you observed.

My first question now would be.... does the same happen for PWM lights?

rianadon, 2022-12-10:
I have PWM lights so have been playing around with the code. Below are some observations I've found with the flag:

Without flag 18 (expected behavior):
1. I configure 5 PWMs for RGB and CW
- The LEDs are turned off at startup
3. I configure only 2 PWMs on channels 0 and 1. I enable flag 8 (I'm doing this because this executes different code inside the color interpolation).
  - The LED is turned off at startup.
  - Clicking "Toggle Light" toggles the light on and off

With flag 18:
1. I configure 5 PWMs for RGB and CW
- Some LEDs turn on at startup
- The board boots into safe mode (possibly due to overcurrent?)
2. I configure only 2 PWMs on channels 0 and 1. I enable flag 8.
  - The LED turns on immediately after enabling flag 18
  - After a restart, the LED turns on immediately
  - Clicking "Toggle Light" does not toggle the light on and off. However, moving the brightness slider still adjusts brightness.

rianadon, 2022-12-10:
Looking through the code I think I know why this is:

RGB mode interpolates to `finalColors[i]` This array is initialized to all 255s, so when the light first boots, all LEDs will be turned on.
When the LED is toggled on and off with "Toggle Light", `apply_smart_light` will write `finalColors[i]` to 0 if the LED is disabled, or the correct color if it is enabled.

However, the 2-PWM alternate CW mode interpolates the brightness to `led_current_value_brightness`. This variable does not depend on whether the LED is toggled on or off, so this is why "Toggle Light" does not work with this mode.

openshwprojects, 2022-12-12:
Thanks. Can anyone else check this PR with both CW and RGBCW lights before I merge?

Bacto, 2022-12-12:
Hi,

Tested on a RGBCW (BK7231T with BP5758D) and a CW (BK7231T with PWM).
It works, there is no default color set anymore. Thank you :)

The negative point is that there is a small delay between powering-up the light and having it fade-in.
When `finalColors[i]` was fill with 255s, the fade-in started immediately.

Is there a way to speed-up the startup command or define the default color (via `finalColors[i]`) on start?

My startup command is this one:
`backlog ; BP5758D_Map 1 0 2 3 4 ; led_dimmer 100 ; led_basecolor_rgbcw #000000FFFF ; led_enableAll 1`.

I tried to improve it by setting color first but the delay seems to be the same:
`backlog ; led_basecolor_rgbcw #000000FFFF ; led_enableAll 1 ; BP5758D_Map 1 0 2 3 4`

GravityRZ, 2022-12-12:
what do you need testing.

i have the same light and never experienced those problems but if have save color settings enabled.
so it is on warm white and when i reboot it stays on warm white

rianadon, 2022-12-12:
It looks like the delay from the startup commands is because the commands run once WiFi has been successfully started.
Whereas the option that saves the color settings will apply these before wifi is set up.

I wonder if as an alternative to making a command that sets finalColors, there could be two startup command options: one that runs before the wifi is set up (for setting LED values) and one that runs afterwards (for mqtt publishing or other wifi-dependent commands).

Bacto, 2022-12-13:
@rianadon interesting. Is waiting for WiFi to start mandatory to accept MQTT and other commands?

Waiting for a better solution, having a second startup commands field seems great to me.
What do you think @openshwprojects?

btsimonh, 2022-12-13:
two startup lines may be a good thing.  note that early.bat is now called early, and autoexec.bat is called after delay (which is still before wifi...).

openshwprojects, 2022-12-13:
Sorry, github automatically closed it with PR merge. The question is - did that PR help? 

Two startup commands are not needed, you already have:
- early.bat
- autoexec.bat

Bacto, 2022-12-25:
Hi,

Sorry my late reply, didn't got time to check this.

I confirm that using the `early.bat` file it works like a charm :)

For those who would have the same problem:

- Do not set a "Startup command"
- Create a `early.bat` file (in "Launch Web Application", "Filesystem", "Create File") and put this in it: `backlog ; BP5758D_Map 1 0 2 3 4 ; led_dimmer 100 ; led_basecolor_rgbcw #000000FFFF ; led_enableAll 1`

Thank you to all of you for the patch and the explanations! 🥰

Adrien

PR #533 (2022-12-13): Fix startup and toggle issues with Smooth LED transitions flag
As I've explained in this comment: https://github.com/openshwprojects/OpenBK7231T_App/issues/532#issuecomment-1345404174, the default initialization of `finalColor` and the failure of color interpolations in the alternate cw mode to take into account whether the light is enabled will cause the light to initially be turned on after boot.

I don't have a build environment set up so I haven't tested these changes, but they should fix #532.
closed 2022-12-25 by Bacto