openshwprojects/OpenBK7231T_App #536: Power Save on T/N

btsimonh, 2022-12-13:
This is a placeholder issues to document findings on power save on T & N

T & N devices have a power save mode, but currently we don't enter it.

Power save mode has been shown to be able to enter power save for up to ~40ms at a time, total of ~250ms per second is the best I have been able to achieve.

Things that need to be done to get power save to work:

In order for power save to kick in, the device must be idle for >2ms.

Things that prevent this:

1/ QuickTick timer.
a/ process buttons from edge interrupts.  Tested and works well?
b/ identify usage of quicktick, and assess/test impact for slowing it down to 50ms cycle.

2/ logging thread(s).
a/ trigger serial log output from pended timer function.  Tested and works.
a/ trigger tcp log output from (the same?) pended timer function.

3/ http logging.
When HTTP logging from the webapp is used, the device will not currently enter power save - do we poll for logs too often?
TODO :- reduce webapp polling rate

4/ IRRecv.
a/ run IRRecv from GPI edge interrupt. (prototyped - BUT... works fine until power save is enabled :(.  When power save is enabled, it can no longer receive reliably - even when holding a button down which keeps in in high power mode, data is still not received well?)

5/ UDP receive: SSDP/DGR
Either:
a/ use netconn connections and callbacks.
b/ create a thread and use blocking sockets.

Please add thoughts as comments against this issue.

Specific help wanted:
Create a generic UDP task which waits on mulitple sockets (using select()) , and calls back generic functions with received data.
SSDP and DGR can then configure this thread dynamically with their UDP sockets on start/stop, and provide callback functions.
Callback functions can 'pend' actions against the timer thread as required, or take immediate actions (bearing in mind multi-thread aspects).

openshwprojects, 2022-12-13:
Hey, what is the underlying reason for the UDP handling change? The current approach - non blocking sockets, checking for recv every quick - is not compatible with power saving?
My first expectation was that we can just check for recv less frequently and it will be enough.

btsimonh, 2022-12-13:
my thoughts are that 
1/ we don't want to miss any packets (and so if polling, must poll fast ->not good for power save).
2/ select() gives us the ability to wake a thread from a receive, so why not use it, this is what it is for....?
3/ udp with callbacks from LWIP is a BIGGER change, especially working out how to enable multicast rx using the raw API...

4/ we should test rx bandwidth on multicast (e.g. spoof DGR packets and log continuity errors), and understand packet loss.

openshwprojects, 2022-12-13:
@btsimonh I remember that in the past I was doing polling every second (only every second) for DGR and it didn't miss packets. That was before I introduced QuickTick for drivers.

btsimonh, 2022-12-13:
We need to get it so there is no room for issues.  We already have a report of DGR missing packets....  i.e. we need to implement it 'correctly'.  (I hate the words 'correctly' and 'properly' - because they imply one person's opinion is more valid than another persons.  But in this case, I do feel polling for UDP packets is flawed, and running a single UDP thread with select() is not so hard; a few hours work.

WiredLife, 2023-01-17:
is the power save feature now available @ ([1.15.339](https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.15.339)?
how i can enable it and will influence the latency when i enable a light over mqtt?