openshwprojects/OpenBK7231T_App #537: Feature Request: Add support for MQTT Light Transitions

bscholer, 2022-12-13:
In Home Assistant, there is [support for transitioning lights](https://www.home-assistant.io/integrations/light.mqtt/#json-schema) from one state to another by using MQTT with JSON. Is there any plan to implement this?

After some quick browsing of the code, it seems like most of the code is already there to do the LERP, and is being used by `Flag 18 - Smooth transitions for LED`. This would just use a custom time instead.

If switching to the JSON MQTT schema is a huge lift, another alternative could be a customizable transition time for flag 18.

mattjacobson6, 2023-06-24:
Wanted to know if this is on the radar for the roadmap.  A lot of progress has been made in the past months.  How feasible would it be to enable this?

roabyd, 2023-07-06:
I'd be very interested in this also. This is the last thing missing from a cost effective light :)