openshwprojects/OpenBK7231T_App #538: Native HomeKit support

tonysprenk, 2022-12-13:
Will it ever be possible to have firmware for these MCU’s to have native HomeKit or Matter support? Like [HAA](https://github.com/RavenSystem/esp-homekit-devices) for ESP8266.

mikimause9, 2022-12-14:
Firstly I don't think there is enough memory on the module to put HomeKit support. The other thing is that this project is mostly focused on utilizing MQTT as it can be connected to most of the devices as HomeKit works just with apple.

tonysprenk, 2022-12-14:
Fair enough. I guess the same counts for Matter support?

mikimause9, 2022-12-14:
With Matter, it is a kinda different story. It is a new protocol, and it is has just raw documentation. To even try we have to wait for somebody to try to port it to a platform like esp8266/esp32.

mikimause9, 2022-12-14:
And again I don't think this microcontroller is powerful enough to support it.

GravityRZ, 2022-12-16:
that's why we use homebridge(as a container)
most stuff not supported by homekit can be used when applied through homebridge

korgoth, 2023-02-16:
> that's why we use homebridge(as a container)
> most stuff not supported by homekit can be used when applied through homebridge

Is there a homebridge plugin to interface with OpenBK firmware? I am unable to find such? I guess i could use HTTP accessory plugin but that doesnt seem very stable to me. 

Also - Tasmota have now introduced experimental Matter support. Do you guys think that this can be somehow merged into OpenBK?

udovic2100, 2023-02-18:
It should be possible to connect OpenBK devices via MQTT to homebridge and from there to AppleHomeKit. As a homebridge plugin you can use mqtt-thing. I'm about to try that with my HomeMagic LED controller.

korgoth, 2023-02-18:
Im trying not to use MQTT. I find it slow and unreliable. The current device i was looking to replace is using MQTT and unfortunately doesnt remember it state on power loss. Thats why i wanted a device that was using WiFi.

crunchtime2k, 2023-03-03:
Same issue here , is there anyway to have homebridge control openbk devices locally and autodiscover without a broker like MQTT?

dinhchinh82, 2023-06-09:
I also expect whether Openbk can be connected to Homekit like the way some ESP8266 library can do?

dinhchinh82, 2023-10-31:
As I know, the ESP8266 is even more weak than beken7231 and ESP8266 still could support native homekit so I hope beken7231 will be supported.
https://github.com/RavenSystem/esp-homekit-devices

tybkyb, 2023-11-03:
Isn’t there already a working example of a matter over WiFi implementation?