openshwprojects/OpenBK7231T_App #541: Led brightness/expo mode default value?

openshwprojects, 2022-12-14:
After merging in https://github.com/openshwprojects/OpenBK7231T_App/pull/539 ,
_/* exponential mode command handler, usage: led_expoMode <0|1|2|3|4>
   exponential modes: 0 = Off
					  1 = 1% min brightness with moderate exponential
					  2 = 1% min brightness with full exponential
					  3 = 0.1% min brightness with moderate exponential
					  4 = 0.1% min brightness with full exponential
*/_
I think we should choose a default value for brightness mode on all devices.

Can you guys check which one in your opinion gives best results? 
Currently, the default value is 0 = Off

WiredLife, 2022-12-16:
https://github.com/openshwprojects/OpenBK7231T_App/issues/546

WiredLife, 2023-01-03:
I updated to current firmware and the expo mode doesnt work as i thought, tried all settings. With https://github.com/openshwprojects/OpenBK7231T_App/pull/447 i can go very down with the brightness.
With the expo mode not.
Very sad

openshwprojects, 2023-01-08:
@WiredLife You have convinced me. We're merging #447 soon. I will allow float values for PWMs. I am closing this issue now, let's continue discussion in #447 .

closed 2023-01-08 by openshwprojects