openshwprojects/OpenBK7231T_App #546: Expo mode at 100% brightness is dim (possibly a BP5758D-specific oddity)

dwildstr, 2022-12-15:
Since 1.15.181, setting my BP5758D bulbs to 100% brightness has made them very dim, while 99% brightness is nice and bright. Executing led_expoMode 0 makes them bright at 100% again, so this looks to be an exponential-mode specific issue, possibly some sort of roundoff weirdness edge case happening at 100% exactly? I haven't studied the brightness-modification code very closely, but it looks like something odd is happening with the calculation on line 361 of `cmnds/cmd_newLEDDriver.c`:
```
final = raw * (powf(expo_base, g_brightness * expo_factor) / expo_factor + expo_offset);
```
because by my calculations, on expo mode 2 with `g_brightness` equaling 1, this multiplier is about 1.009; with `g_brightness` of 0.99, it comes out to 0.966. Are greater-than-1 effective brightnesses becoming confusing when passed to the BP5758 driver (I imagine straight PWM doesn't have a problem with them at all treating a greater-than-100% duty cycle as just always-on, so maybe this bug doesn't manifest with PWM configs)?

It seems like there are two places where this wants a fix for best practices --- expo mode shouldn't be outputting relative brightnesses of more than 100% at all, and LED drivers shouldn't be accepting them uncritically (presumably pulling values  more than 100% down to 100%, and values below 0% up to 0%).

WiredLife, 2022-12-16:
I can confirm this bug on LSC GU10 RGB with SM2135

openshwprojects, 2022-12-16:
[image]
I can just add a clamp, wait

openshwprojects, 2022-12-27:
Clamp was added, is it solved now? I can assume it's solved? Tell me if somethings wrong and I will reopen.

closed 2022-12-27 by openshwprojects