openshwprojects/OpenBK7231T_App #551: W600 instructions missing

Tryfos, 2022-12-18:
I'm pretty sure that the WiKi used to have instructions to build/flash on W600/601, but now this seems removed. Has something changed in relation to W600 devices?

openshwprojects, 2022-12-18:
I am not aware about any chances. Which wiki page are you referring to?

Tryfos, 2022-12-18:
> I am not aware about any chances. Which wiki page are you referring to?

Here https://github.com/openshwprojects/OpenBK7231T_App  , if someone scrolls down, there are instructions for flashing every chip supported except W600. I think there was a section about W600 (how to flash and the recommended tool). Now I can't find one. Here https://github.com/openshwprojects/OpenBK7231T_App/wiki/Wiki-Home  **W600** is not even mentioned as supported.

openshwprojects, 2023-01-29:
https://github.com/openshwprojects/OpenW600

closed 2023-01-29 by openshwprojects