openshwprojects/OpenBK7231T_App #552: Support for Power Meter?

deanfourie1, 2022-12-21:
Just curious, are these supported?

https://www.aliexpress.com/item/1005004967717030.html

Chip reads Model: CBU

Thanks

openshwprojects, 2022-12-21:
We have support for similiar meters, but Idk about this precise model.
But we can just help you getting it supported, if you have one. Create a topic on our forums and we will guide you step by step.
Just like here:
https://www.elektroda.com/rtvforum/topic3928897.html

deanfourie1, 2022-12-21:
Thanks, have posted. I believe they are based on the BK7231N chip.

closed 2022-12-21 by openshwprojects

jf-wm, 2023-01-08:
It uses a BL0942 powermeter IC inside. Unfortunately the IC is connect in SPI mode with the BK7231N, therefore I couldn't get it to work with OBK with the original wiring. Instead I rewired the the connection on the PCB to UART, that's works quite okay. Nevertheless the module still has dropouts and doesn't deliver any data via MQTT for some hours every few days. I could figure out what problem causes this.