openshwprojects/OpenBK7231T_App #553: Support for WPA key as 64 hex digits

s09bQ5, 2022-12-21:
**Describe the bug**
The WPA2 standard (IEEE 802.11i) specifies a method for converting passphrases into the 256 bit PSK. The standard assumes that this method is used only for strings up to 63 characters to be able to distinguish the case where the PSK is passed directly as 64 hexadecimal digits. 

OpenBeken so far does not support passing the PSK as hexadecimal digits. If you enter more than 63 characters, the rest is silently dropped. I checked the BK7231N SDK and there appears to be code to handle the 64 hexadecimal digits case, but the intermediate arrays to hold the key are one character too short.

**Firmware:**
- Version 1.15.215
- Chip/model: BK7231N

**To Reproduce**
Steps to reproduce the behavior:
1. Enter a PSK as 64 hexadecimal digits
2. The device fails to connect to the network
3. Go into safe mode by quickly repowering the device several times
4. Go to the wifi config menu
5. See that the last digit in the passphrase field is missing

**Additional context**
There is no passphrase for my wifi that I could enter instead of the 64 hexadecimal digits. The PSK consists of 256 random bits. So either this is fixed in OpenBeken or I have to change the passphrase on all of my devices.

openshwprojects, 2022-12-21:
The current password field lenght is 64 bytes, which also must include the NULL terminating character for the screen.

By 64 hexadecimal digits, you mean... the hex notation? Where two chars make one bytes? FFAABB? 

Then we would need to make our field twice as large and also include space for NULL terminator....

Or do you want to pass field to be just one byte longer?

Hmmm our config has no free bytes directly after pass, we would need to make a config upgrade function to convert between old config and a new one....

@btsimonh , thoughts?

s09bQ5, 2022-12-21:
Only one byte more is needed. 256 bits = 32 bytes = 64 hexadecimal digits.
It would be possible to store the information in the existing space (by converting the hex string to binary and filling the rest with something != 0), but then the simplicity of the code is lost.

openshwprojects, 2023-01-12:
Solved with pull request submitted by user. Thanks!

closed 2023-01-12 by openshwprojects