openshwprojects/OpenBK7231T_App #554: SM2135 never turns off RGB leds (even in C/W mode)

JesseMcL, 2022-12-22:
Hi,

I have just tried your app for the first time. It is great to have a tuya alternative!

I have Arlec GLD132HA (Arlec Smart LED Globe RGB+CCT 12W, 1050 lumen).
It uses WB2L and SM2135E**J**.

SM2135EJ is broken from 1.15.206 onwards (EDIT, 1.15.205 works)

I checked the pinout and was able to configure with:
 - P24 (PWM4) SM2135CLK
 - P26 (PWM5) SM2135DAT

Running `SM2135_Map 1 2 0 4 3` causes the correct RGBCW colours to display

I have tried with and without all combinations of:
- Flag 4 - [LED] Force show RGBCW controller (for example, for SM2135 LEDs, or for DGR sender)
- Flag 9 - [SM2135] Use separate RGB/CW modes instead of writing all 5 values as RGB

I note flag 9 functionality appears to have been commented out from 1.15.206

`SM2135_RGBCW 0000000000` only turns off warm/cool white channels. R/G/B remain at their previous value
Setting brightness to 0 has no effect on RGB, light remains at previous value.
Toggling light power (on off) also has no effect on RGB, light remains at previous value.

The tuya documentation says that SM2135EJ (and EH) can support RGB and CW at same time.

I've found a full chinese datasheet for the SM2135EJ here: http://www.chinaasic.com/chipDetails/detail_351.html, but its images not text so difficult to translate. I've attached a PDF print of the images though.
[SM2135EJ_cn.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/10284264/SM2135EJ_cn.pdf)
It appears that Byte2 is a very different function to the SM2135E, there is no register (B7) to select RGB/WY and instead only a standby enable (B[5] = 1).

In this case, the example (even in chinese) in that datasheet shows that you should just be able to write all 5 RGBCW (RGBWY) registers instead of alternating between writing RGB and CW, and has a standby mode with Byte2[5] instead of the Byte2[7] RGB/WY mode.

Seems the flag OBK_FLAG_SM2135_SEPARATE_MODES is meant to allow both behaviours, however was broken (commented out and defaults to as though the flag was applied, from 1.15.206 onwards.

SM2135EJ could also write 0b00100000 to byte 2 (instead of SM2135_RGB) when 'off' to enter standby, with RGBWY = 0x0000000000.

I realise i've now solved my own problem with downgrade, but this regression should be fixed in new version.

openshwprojects, 2022-12-22:
Thank you, for now, I have restored OBK_FLAG_SM2135_SEPARATE_MODES. 
If you want to propose a better solution, create a pull request.

I have recently spoken with one of the SM2135 users and there was a suggestion that the flag is not needed, which I accepted, while sadly being unable to test on this SM2135. Turns out I was wrong.

The OBK_FLAG_SM2135_SEPARATE_MODES was added because on some SM2135 only RGB channels were working. It seems your post explains this behaviour well. There are two SM2135 kinds, although I also think it might be related to the incorrect current setting, maybe in "write all RGBCW mode" the bit shift for CW current limit is missing and CW current is set 0 - I am not sure, I remember reading something like that.

Does the newest update with alternate mode revert solves your problem? At least for now, then we can do a better solution

JesseMcL, 2022-12-28:
Hi, thanks for the update and I can confirm it works with OBK_FLAG_SM2135_SEPARATE_MODES flag OFF. If i can get the toolchain set up then I might try submit a change request for standby mode for the ICs that support it. but for now it is great, thank you!

closed 2022-12-28 by JesseMcL