openshwprojects/OpenBK7231T_App #555: BK7231N Smartsocket does not connect to Wifi

btmerz, 2022-12-22:
Hello,
I finally managed to get OpenBK installen on my BK7231N smart socket.
I was also able to get  the module configered, so that Relay, LED, pbutton and power measuring work.
But unfortunately I'm not able to get the device connected to my WIFI. SSID (FRITZ!BOX 7490) is set correct and I also checked my wifi password several times. But It doesn't show up in the router and the WIFI LED is constantly blinking.
When ich restart it in save mode I can connect to it in AP mode without problem. But i need it in my network for MQTT communication.
Do you have any advice?

Thanks for the great project!

**Firmware:**
- Version 1.15.215]
- Aubess 20 Smart socket with power monitoring
- Chip/model: BK7231N?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2022-12-22:
Would it be possible to see UART log from the device?

Are you able to compile the minimal WiFi connect example from our N SDK and see if it also fails to  connect?

btmerz, 2022-12-22:
Sorry, but I'm new to this project. How do I get this log? And is there a tutorial, how to compile this minimal version? Thanks for your help...

openshwprojects, 2022-12-23:
Log is available at the TX2 pin of your BK device. You should connect UART directly to get that, so it might be problematic.

Regarding minimal build... well, I can see how that can be problematic. I would need to copy T SDK project example to N SDK for you. I might look into that later.

Anyway, is it your only Beken device? Are you able to tell us whether it is a problem specific to your device or specific to your WiFi? Does it connects to other WiFi AP well? That would help nail down the issue.

sebastianheierhoff, 2023-02-17:
Hello,
any chance that your Wifi password contains special characters? I am also using a Fritzbox 7490 which can create two SSIDs (one for guests). For testing, I set a very simple wifi password on the guest SSID to which the BL602 with the current firmware can connect just fine. When switching over to my main SSID with a more complex password, the module (BL602) does not connect and gives a Wlan802.11 status code = 53 (Invalid shared key (pairwise master key identifier or PMKID)). I set the Wifi password via MQTT, web requests & webUI console multiple times and am 100% sure that it is correct. Interestingly, the password shown in the logs is correct, as well. It does, however, show Mac 00:00:00:00:00:00. Is that correct?
Thank you

stefan73, 2023-11-01:
Same here with a Fritzbox 7560. Wifi connection cannot be established. See also serial log below. I can exclude the wifi password characters. I use the same wifi password on a cheap china wifi repeater. The BL602 with this flashed firmware connects here w/o problems. With the original firmware from ewe the device connected to the Fritzbox, so it is a firmware issue that can be solved.

Serial log sniplet:

[WF][SM] Retry Again --->>> retry connect
[WF][SM] Exiting disconnect state
Delete Timer.
[WF][SM] State Action ###disconnect### --->>> ###connecting###
[WF][SM]  Action Connect
           ssid *** my network name removed ***
           ssid len 8
           psk *** my wifi password removed ***
           psk len 24
           pmk 
           pmk len 0
           mac 00:00:00:00:00:00
connecting using vif_idx 0
===start sm_get_bss_params===
bssid[0] = 0xffff
bssid[1] = 0xffff
bssid[2] = 0xffff
search ssid =  *** my network name removed ***
result ssid index = 0
===end sm_get_bss_params===
[WF][SM] Entering connecting state
[APP] [EVT] Reconnect 84804
[APP] [EVT] Reconnect 84804
[APP] [EVT] Connecting 84805
[APP] [EVT] Connecting 84805
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:1
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:1
Probe Response
timestamp = 0x24014a79991
wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:1
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:1
Probe Response
timestamp = 0x2402219bf80
[11:57:40.481] - wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:1
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:1
[11:57:40.591] - wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:1
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:1
------ 247 set default key 0x42020f10, key ptr 0x00000000
[11:57:40.917] - [APP] [EVT] SCAN On Join 85320
[APP] [EVT] Unknown code 10, 85321
Association fail
ure, Wlan802.11 status code = 53
ind ix 0x42029164, chan_ctxt is 0x42021dfc
connect fail

sebastianheierhoff, 2023-11-03:
Does it connect to your Fritzbox if you change the PSK to something "easy", i.e. without special characters? Mine works fine if I change my key to "test" but doesn't if I use a longer key with special characters. Thus my assumption that it has something to do with the Fritzbox and its behavior when using complex keys.