openshwprojects/OpenBK7231T_App #557: Spectrum Nymphea light and some celing light

bart-sofomo, 2022-12-23:
Hello,
I bought light on:
1. https://spectrumsmart.pl/pl/p/Oprawa-LED-Wi-FiBluetooth-NYMPHEA-24W-1840lm-CCT-DIM-230V-130st-IP20-390x70mm-BIALY-okragla/87 -> Spectrum Nymphea 24W 1840lm CCT
2. https://allegro.pl/oferta/plafon-sufitowy-led-tuya-wifi-18w-ze-sciemniaczem-12709325490?snapshot=MjAyMi0xMS0yMlQwNjo0NTo0NC4xODlaO2J1eWVyOzkzZjI4Y2U3Zjg4M2QzYTA0NGIzM2YxOWQwZDQ1NWIzMmQ5YzE5OWE0MWMwZmY0MzY1NjBmZTEyNDA5ZGZkNDA%3D -> Smart Led Moon 18W 1260lm CCT

I setup pins like:
1. 24: PMW1, 26: PWM2 with Flag 8 for Spectrum
2. 6: PMW1, 7: PMW2 for Led Moon

I have however 2 problems:
1. light seems to do not get a full brightness on Spectrum - it seems like it doesn't get full potential. 
2. Warm/Cold are inverted.

Do I have to specify some additional commands in order to make it working properly?

openshwprojects, 2022-12-23:
Hello, I think you can try those things:
1.  Maybe you need to enable 600Hz PWM on Beken? Flag 11? There is such options in the flags, but that's a guess for now.
2. You can easily use PWM_n role for that purpose, I think. it's like PWM but inverted. Should be seamlessly integrated into LED driver.

By the way, if you have a not listed device here:
https://openbekeniot.github.io/webapp/devicesList.html
Please submit a teardown/review to our forums, along with the template, so other users can benefit.
(If you are from Poland, you can get a free SD card for submitting each article on our Elektroda forums, if from EU, then not sure, I would have to ask admin where he can ship)

bart-sofomo, 2022-12-24:
@openshwprojects thanks - created thread of forum :)
1. The 600Hz did not work but I do not really care about like full potential of the light, it's to bright anyway :)
2. Inverting pins helped - more will be in the forum threads.

closed 2022-12-24 by bart-sofomo

openshwprojects, 2022-12-24:
@bart-sofomo for the sake of completness, if you are a technical person, we could look into the point 1 deeper, but it might require access to scope or ability to compile obk with different PWM frequencies. I am currently not sure why the range wouldn't be full.

Our PWMs are from 0% to 100% . So it has to be something else. 

Maybe can you tell us the names of the ICs on the first device and we will look up the datasheet and just check if it works with 600 or 1kHz frequency PWM?