openshwprojects/OpenBK7231T_App #560: [bug] flag 8 only works when 2 pwm channels set, fails when 5 set RGB(color temp)(white brightness)

heathfx, 2022-12-24:
**Describe the bug**
flag 8 only works when 2 PWN channels set, fails when 5 set RGB(color temp)(white brightness).

I have a 100w Cree CMA21-100W-AL-9ACK with RGB and W+Color Temp.

P7 (PWM1) controls the brightness of the white light.
P8 (PWM2) controls the color temperature, 100% duty cycle is full warm, 0% is full cool.

The RGB channels are controlled with 3 PWM channels (pretty typical).

**expectation**
The expectation is that when enabling flag 8 with 5 PWM channels, the behavior on the 3 RGB channels remains unchanged, while the 2 channels controlling white light switch from controlling warm white on ch 5 and cool white on ch 4
-to-
controlling color temp on ch 4 and white light brightness on ch 5.
on channel 4: 100% duty cycle is full warm, 0% is full cool.

**reality**
Absolutely nothing changes when flag 8 is enabled when more than 2 PWM channels are present.

**Firmware:**
- Version 1.15.223
- Device? Cree CMA21-100W-AL-9ACK
- Chip/model: BK7231T (tuya WB8P)
- Device config? 5 PWM channel set, flag 8 enabled

openshwprojects, 2022-12-24:
I think you are correct, thank you for reporting, but that's not a bug. It was simply not implemented.
I can try doing that for you today, will you help with testing?

heathfx, 2022-12-24:
Thank!

I would be happy to test today. (Off work for Christmas holiday, good time to tinker.)

openshwprojects, 2022-12-24:
Ok I am waiting for your test resul;ts

heathfx, 2022-12-24:
I did an OTA update to 1.15.226 and the white light function now works as expected with flag 8 enabled, but when I choose an RGB color, channel 5 is left on causing the bulb to have both the RGB and white leds on simultaneously. In RGB mode, channel 5 should go to 0.

openshwprojects, 2022-12-25:
Thank you for testing.
I have added a fix, can you check now?

heathfx, 2022-12-25:
Works perfect now. Thanks!

Now time to solder some wires to the rest of my bulbs and flash them!

heathfx, 2022-12-25:
Well...almost. Channel 4 should always mirror the color temp slider (last setting). If you turn the bulb on and the color temp is anything other than 6500K it will flicker once at power on. When in rgb mode or turned off, ch 5 should go to 0 and ch 4 should remain at its previous setting (color temp).

Channel 4 needs to be set before applying power to the white LEDs (channel 5), otherwise you get a glitchy look to the power on (like a florescent lamp turning on).

openshwprojects, 2022-12-25:
There are multiple ways to fix that, but can you try that?
https://github.com/openshwprojects/OpenBK7231T_App/commit/96d70fdf95ab08ff25b941d757a8460b58844410

heathfx, 2022-12-25:
This latest build works perfectly!

Thanks a bunch for your help!

openshwprojects, 2022-12-27:
I think it's safe to close it now.

closed 2022-12-27 by openshwprojects