openshwprojects/OpenBK7231T_App #561: MQTT initial reported IP address not updated

NielsPiersma, 2022-12-24:
**Describe the bug**
After running "Start Home Assistant Discovery" the CURRENT IP address is reported to HomeAssistant.
When the IP address changes, this is not reported to HomeAssistant. The change can be on the same or any other subnet.

**Firmware:**
- Version [e.g. 1.15.188 to 233/latest]
- Device LSC Smart Plug
- Chip/model: BK7231T
- Device config: n/a/

**To Reproduce**
Steps to reproduce the behavior:
Goto home assistant configuration
Run start home assistant discovery
move to a different subnet, but keep using the same HA server
Check home assistant MQTT device and see the ip address reported in the status page.

**Additional context**
Quick fix is deleting the device from HA and restarting HomeAssistant Discovery.

Niels

openshwprojects, 2022-12-26:
Thanks. I think you are indeed correct. IP is broadcasted on MQTT connect and not later. If it changes during runtime, it can indeed not be broadcasted to MQTT again.

I will look into it, I will also broadcast IP when it changes.

openshwprojects, 2022-12-26:
@NielsPiersma , I have added a fix, but I don't have dynamic IPs - I am using DHCP reservation for MAC.
Can you check if it works for you?
https://github.com/openshwprojects/OpenBK7231T_App/commit/adcf2ef0016b9fc5f00bb3a506d03c5281b9fcb4
It basically checks IP every frame and if it changes, it does MQTT Ip broadcast.
Thanks!

NielsPiersma, 2022-12-26:
Afternoon,

 

Not working as expected at this moment. I flashed device but only after starting the MQTT Discovery the ip addres was updated (to .99)

 

So I retested.

 

Current IP: 10.4.15.99 (from device)

 

In home assistant:

 

I updated the dhcp reservation, invalidating 10.4.15.99, so it must get a different IP address;

 

Now we have 10.4.15.102.

 

And over to HomeAssistant

 

 

Still stuck at 15.99

 

Re-run discovery from the device:

 

 

And updated in HomeAssistant…

 

 

Can we trigger an update / rediscovery? Or maybe do it on some kind of regular basis?

 

Niels

 

 <https://github.com/openshwprojects/OpenBK7231T_App/commit/adcf2ef0016b9fc5f00bb3a506d03c5281b9fcb4> adcf2ef

NielsPiersma, 2022-12-26:
I think all my pictures were dropped ;(.

Any way,
[image]
Also MQTT explorer gives it away... After the re discovery it is updated towards HomeAssistenant.

NIels

openshwprojects, 2022-12-26:
The current modification just does the publish at "IP" topic of MQTT when IP changes, are we referring to the same thing? 
@iprak  what do you think? It looks like related to HA discovery

NielsPiersma, 2022-12-26:
Hi,

When I do a publishall it updates the ip address. So I am considering adding it to autoexec.

Niels

NielsPiersma, 2022-12-26:
> Device should currently publish IP automatically on change:
>  <https://user-images.githubusercontent.com/85486843/209566136-8a639784-f404-430b-a710-adbbd2f93b71.png>
> so the question is why this solution doesn't work for you.... do you need anything more than this publish?

I think it indeed does publish the change when powered on, but not when the device is restarted, and has a new IP address is assigned.

Maybe the behaviour is different between a lease renew when powered on vs renew during startup.

I’ll investigate this by changing the lease time to a few minutes and see if the lease is update without restarting the unit.

 

I’ll also check if the publishall in the autoexec resolves the issue.

 

Thanks 

 

Niels

NielsPiersma, 2022-12-26:
Okay,
Tested, but still not working (in HomeAssistant). However, it may be related to HomeAssistant and not to OpenBeken.
Well the packet trace from the DHCP server. As you can see after a few NAK's for 10.4.15.102 it does an ACK on the 10.4.15.99.
[image]

And when looking into MQTT Explorer it has updated as well.
[image]

But in HomeAssistant is still is using the "old" address.

[image]

Keep in mind that switching from inside HomeAssistant is working fine, it is just the IP address stored in the configuration.

After executing the custom command "publishall" from the device 

[image]

We see the data isn't updated in HomeAssistant as well. Still the former IP address;

[image]

After a reboot, still no update of the IP address in HomeAssistant.

However, when deleting the device from HomeAssistant and executing Start HomeAssistant Discovery, it is reported with the correct IP address.
[image]

Based on this testing IP changes during operation and after a reboot are currently not reported back into HomeAssistant.

I can do more testing if needed.

Niels

openshwprojects, 2022-12-26:
I am starting to think that the HA is not listening to the IP publish and the thing I added does not relate to your request.

You are referring to the AutoDiscovery, which is done just once, and also might contain IP.

I must ask @iprak  about it.

PS: If you listen to MQTT publishes done by that device, when a DHCP changes IP, does it publish the new IP?  I bet it does, it's just that's not what you're asking for.

NielsPiersma, 2022-12-26:
@openshwprojects ,
"PS: If you listen to MQTT publishes done by that device, when a DHCP changes IP, does it publish the new IP? I bet it does; it's just that's not what you're asking for."

I fully agree with your bet; I assume the IP address **is** updated. MQTTExplorer does report the new address instantly.
[image]
 
Likely there is something else published to HomeAssistant, (initially) that needs to be updated. My tasmota's are not running MQTT anymore, but as a native tasmota device, I can't test MQTT. I'll look for another MQTT device and see if we have the same behavior.

iprak, 2022-12-26:
HA discovery is manually invoked.

So if the ip address changes, then `Device Info`  will continue to show previous address. The HA integration if through MQTT so the integration should still work. Unfortunately, the entire discovery package has to be sent to correct this. We could try re-invoking discovery on startup and on ip-address change.

openshwprojects, 2022-12-26:
We already have ability to call AutoDiscovery from console command, maybe just schedule autodiscovery when IP changes?

But not all of people are using that... and it might not be desirable by default?

NielsPiersma, 2022-12-26:
I can confirm @iprak device info statement. The IP address is published to 
***homeassistant/switch/OpenBK7231T_BF512F1E_relay_1/config***
[image]

So that needs updating when the IP addresses changes.

NIels

openshwprojects, 2022-12-26:
@NielsPiersma  maybe adding an event for IP change would help? Then, in autoexec.bat, you could script the AutoDiscovery to happen again automatically...

but I am not sure if such user-scriptable solution is suitable here.

NielsPiersma, 2022-12-26:
> We already have ability to call AutoDiscovery from console command, maybe just schedule autodiscovery when IP changes?
> 
> But not all of people are using that... and it might not be desirable by default?

When doing ipv6 the only realistic option is dhcp... Sooner or later something that is likely needed.
NIels

NielsPiersma, 2022-12-26:
> @NielsPiersma maybe adding an event for IP change would help? Then, in autoexec.bat, you could script the AutoDiscovery to happen again automatically...
> 
> but I am not sure if such user-scriptable solution is suitable here.

For me, that is fine for now. I understand where the issue is. 

But changing IP addresses is not something unheard of ;)
NIels

NielsPiersma, 2022-12-26:
@iprak how can I do AutoDiscovery from the command line?
And would it be possible triggering it from HTTP://x.y.x./?cmnd+..... and as a MQTT cmnd topic payload?

I know many requests :) but I am managing over 1000 Tasmota devices and provisioning is the only wat managing them in a controlled way.
NIels

iprak, 2022-12-26:
Can I ask, what are you using the url in Device Config for? I would have expected that addresses to be provisioned in such a large setup based on mac address.

NielsPiersma, 2022-12-26:
This is how we mass reconfigure initial hosts in AP mode. We used tuya cloud-cutter and scripted the process. 
http://192.168.4.1/cm?cmnd=backlog+SetPinRole%206%20LED_n;SetPinChannel%206,1;SetPinRole%207%20Btn;SetPinChannel%207,1;SetPinRole%208%20Rel;SetPinChannel%208,1;SetPinRole%2026%20wifiled;SetPinChannel%2026,0;SSID1%20[wifissid];Password1%20[wifipwd];MqttUser%20[mqttuser];MqttPassword%20[mqtt-pwd];MqttHost%20[ip of ha];restart

Niels

iprak, 2022-12-26:
I wonder if more sustainable solution might be look into the router settings. DHCP usually assigns the same ipaddress for a previously connected mac but the behavior is not consistent across DHCP implementations. 

But triggering `scheduleHADiscovery` should also work although you need to build some process to trigger it via listening to MQTT message.

openshwprojects, 2022-12-26:
i have added IPChange event (with argument 0 - it's just a stub for syntax). The event is cast when IP changes. 

**addEventHandler IPChange 0 scheduleHADiscovery**

openshwprojects, 2022-12-29:
Any news on this? @NielsPiersma

NielsPiersma, 2022-12-29:
Sorry recovering from flu

NielsPiersma, 2023-01-01:
@iprak , I am testing now, just to confirm, do I need to add 'addEventHandler IPChange 0 scheduleHADiscovery" somewhere (as a startup command?) Or is it enabled by default?

Just flashed version 268, will try now without any changes.

NIels

NielsPiersma, 2023-01-01:
Evening @iprak 
I am starting with OpenBK firmware 268. I just updated to this version and did not (yet) issue an MQTT rediscovery from OpenBK.
 
[image]

@ipark, adding the current IP address on the about page would be helpful. If needed, I can open a new change request for that.

Current IP Registration in DHCP server (free lease, not pinned/reserved)
 
[image]

And HomeAssistants’ view on things.
[image]
 
As you can see, it nicely points to the correct, current IP address; also, notice that the firmware revision is not updated either; it reports version 230 and not the just flashed 268. @iprak, can we do a rediscovery as well after a firmware update?
Now I’ll invalidate the IP address and reboot the unit, so OpenBK needs to acquire a different address.

[image]
Reservation with a different MAC address 😊

Now reboot the plug and see the results.

We’ve migrated to 10.4.15.102
 
[image]

Let’s see HomeAssistant:
 
[image]

No change; still has the old firmware version and the old IP address.

Executed the custom command; "addEventHandler IPChange 0 scheduleHADiscovery"
[image]

 
First, updated HA with the new information by starting HA discovery:
HA information after triggering HA rediscovery:
 
[image]

As before, after manually running HA rediscovery, the information in HA is correct.

Now triggered, another IP change by forcing 10.4.15.99 to this mac address.
 
[image]

I am rebooting again to see if HA rediscovery is triggered.
 
[image]

And back she is.
Now see if HA is updated.
 
[image]

Nope, also after 3 minutes still the same. 
Maybe “addEventHandler IPChange 0 scheduleHADiscovery” only works when IP is changes during running operation and not during boot.

I think it would be a good idea to issue an MQTT rediscover on regular basis making sure the information in MQTT is accurate.

Hopefully you can help out resolving the issue

Niels

iprak, 2023-01-02:
Using "Execute custom command" will just run that for that session, try setting the same as the "Startup command".

NielsPiersma, 2023-01-02:
@iprak ,
Yes, I'll give that a shot.

Niels

NielsPiersma, 2023-01-02:
I can confirm putting scheduleHADiscovery 10 as a startup command resolves the issue as expected.
Kudos!
Niels

closed 2023-01-02 by NielsPiersma