openshwprojects/OpenBK7231T_App #562: Flashing BL602 does not always  bring up an AccessPoint

sverd, 2022-12-25:
I obtained 2 identical smart plugs with energy monitoring options [AliExpress shop(](https://nl.aliexpress.com/item/1005004988124386.html?spm=a2g0o.order_detail.order_detail_item.3.7ec46d76mwTfpm&gatewayAdapt=glo2nld). After opening them up it turned out to be based on a BL602 board and a BL0937 energy monitoring IC.
After flashing with firmware 1.15.228 the first plug presented an AccessPoint as expected. The second plug however did not bring up an AccessPoint even after a number of flashing attempts that seemed to flash just fine..
Is this a known issue, or did I do something wrong (although the flash seemed to suceed every time)?

**Firmware:**
- Version 1.15.228
- Ëlivco? Model RMC004, Smart Plug with energy monitor
- Chip/model: BL602 with BL0937
- Device config to be determined

openshwprojects, 2022-12-26:
We are not aware about the BL602 plug yet, can you submit us a teardown (and if possible, the pin configuration) here?
https://openbekeniot.github.io/webapp/devicesList.html
You are the first one with such plug.

I have tested multiple BL602 devices and dev board and had no such issue.

**I need to see the serial log from BL602 showing what happens exactly. Can you connect UART and provide one?**

sverd, 2022-12-29:
Thanks for the reply!
I did submit a teardown (or at least I tried, please let me know if it is missing things) together with the results of following the traces on the PCB and checking continuity with a multi meter.

I also tried to obtain the serial log from the BL602. However, I "over-estimated" my soldering skills and ripped some soldering pads in the process...... As a result I'll need to solder even closer or at the IC which I'm not sure I can do. I will give it a try but that will take some time. In the mean time I'll focus on the other plug that flashed fine but only had some trouble with the BL0937 energy monitoring....