openshwprojects/OpenBK7231T_App #563: Best way to determine the configuration of a BL602 device?

sverd, 2022-12-25:
After flashing the firmware sucessfully it also has to be configured. As my device did not seem to be present in the database, I tried to simply follow the pcb traces to the actual pins on the bl602 IC. At that point I got stuck. The number of pins on the BL602 IC is larger than the number of available "P's" in the firmware.
How should pin numbers be translated?  Is there some kinf of scheme or so?

**Firmware:**
- Version 1.15.128]
- Elivco? Smart plug with energy monitor
- Chip/model: BL602 + BL0937
- To be determined

openshwprojects, 2022-12-26:
Hello, thank you for your message.
It seems that you have a BL602 version of BL0937 plug? We have never encountered such combination before. You are the first one.
I am not sure if BL0937 is enabled on BL602 yet. We would need to look into it.
Futhermore, can you show screenshots which pins are missing?
I don't have BL602 plug device but I have a development board.

sverd, 2022-12-29:
Hello, 
thank you for the reply!
I added a "tear-down" with more detailed info. Because I did get the led, relay and button to function, I guess the numbering corresponds with the GPIO numbers for each pin. The BL0937 does however not show up/give any result, so maybe it is like you say and it is not enabled on BL602?
Hopefully the tear down is usefull, otherwise please let me know!