openshwprojects/OpenBK7231T_App #564: NTP TimeZone does not honour +HH:MM

jkwim, 2022-12-25:
NTP Time Zone settings ignore the minute part of offset.

**Firmware:**
- Version [e.g. 1.15.215]
- Chip/model: BK7231N

**To Reproduce**
`ntp_timeZoneOfs +05:30
`

```
ntp_info
--
Info:NTP:Server=192.168.1.235,   Time offset=18000

```

openshwprojects, 2022-12-26:
Hey, the syntax is currently just an integer - a number of hours to shift. 
Would you like us to support more advanced syntax?
Btw, here is full list of commands: https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands.md

jkwim, 2022-12-27:
It is not correct to assume all countries have time zones with offsets only in hours eg: XX:00.

There are many countries with time zones with offsets of XX:30.

[https://en.wikipedia.org/wiki/List_of_tz_database_time_zones](url)

I tried with offset in seconds which did not work obviously.

It would make sense to make a modification to code to accomadate all time zones in the world.

Thanks

openshwprojects, 2022-12-27:
I will look into it and try to support Tasmota syntax:
<html>
<body>

Timezone | -13..+13 = set time zone offset from UTC in hours-13:00..+13:00 = set time zone offset from UTC in hours and minutes99 = use time zone configured with TimeDst and TimeStdUse the Tasmota time zone table to find the commands for your time zone.
-- | --

</body>
</html>

openshwprojects, 2022-12-27:
Done: https://github.com/openshwprojects/OpenBK7231T_App/commit/ab98793619696700a2f2581b3c0b5f7f8a90d18b

jkwim, 2022-12-28:
Tested with 1.15.248 and it works correctly now.

However I find that periodic time reporting in to the console log does not work anymore.

openshwprojects, 2022-12-28:
Thank you for testing. Can you provide more details, what do you mean? What kind of reporting? I don't remember changing anything like that?

jkwim, 2022-12-29:
Here is the message:

```
Info:NTP:Seconds since Jan 1 1900 = 3880991585
Info:NTP:Unix time  : 1672002785
Info:NTP:Local Time : 2022/12/25 21:13:05
```

openshwprojects, 2022-12-29:
Thank you, please keep posting new suggestions. What else can we add?

closed 2022-12-29 by openshwprojects