openshwprojects/OpenBK7231T_App #566: Feature Request :  Detect WiFi connection/disconnection as Events

jkwim, 2022-12-27:
I see following log entried during booting:

`Info:MAIN:wl_status 3
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING
Info:MAIN:wl_status 10
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED
Info:MAIN:wl_status 11
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED
`

Can these statuses be trapped as event handlers so that they can be used in a script?

I would like a script to detect when WiFi is connected and when it is disconnected subsequentlly.

openshwprojects, 2022-12-27:
Good idea, here it is: https://github.com/openshwprojects/OpenBK7231T_App/commit/0f555f5d4e904ee937413ee8f0f6466b3cbc227c

jkwim, 2022-12-28:
Thanks for that quick response!

So what should be enumeration values that can be picked up by a script?

In the above log entries what do the status  3,10,11 mean?

```
enum HALWifiStatus {
typedef enum HALWifiStatus {
	WIFI_UNDEFINED,
	WIFI_STA_CONNECTING,
	WIFI_STA_DISCONNECTED,
	WIFI_STA_AUTH_FAILED,
	WIFI_STA_CONNECTED,
	WIFI_AP_CONNECTED,
	WIFI_AP_FAILED,
};
```

Oh! BTW shouldn't it be `HALWifiStatus_t` instead of  `HALWifiStatus`?

openshwprojects, 2022-12-28:
@jkwim  this is because we have an abstraction layer between SDKs and we are using our own enum instead of BK one, so it's multiplatform and clean. I have added display of enum integer values to "Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING" messages, wl_status is BK-specific and can be ignored.

jkwim, 2022-12-29:
Could you help me to write a script to catch WiFi Connected and WiFi Disconnected events in the following syntax please?
`
AddEventHandler <event> ...`

jkwim, 2023-01-01:
I tried something like this but does not work suggesting that the keyword for the event may not be correct.

`AddEventHandler WiFiState 4 Backlog setPinRole 7 LED_n; setPinChannel 7 1
`
Please tell me the correct syntax

jkwim, 2023-01-01:
Tested on 1.15.266 but no success.

It looks like though event handler has been implemented the command has not been defined

`Error:CMD:cmd WiFiState NOT found (args )
`

openshwprojects, 2023-01-01:
Hello, this message: "Error:CMD:cmd WiFiState NOT found (args ) " means that OBK thinks that "WIFIstate" is a command.
How are you executing that command?
It seems you are doing it incorrectly.
Also, you should use a change handler, I think.

AddChangeHandler WiFiState == 4 echo "wifi state became 4"
AddChangeHandler WiFiState != 4 echo "Wifi state is no longer 4"

openshwprojects, 2023-01-03:
Hmm not yet but I added a lfs_append [filename] [message] command for you, maybe you can use it to debug?
https://github.com/openshwprojects/OpenBK7231T_App/commit/2517cd332729c6432d00de9b3dc7af0491feda46
I will look into it more tomorrow

jkwim, 2023-01-03:
> Hmm not yet but I added a lfs_append [filename] [message] command for you, maybe you can use it to debug? [2517cd3](https://github.com/openshwprojects/OpenBK7231T_App/commit/2517cd332729c6432d00de9b3dc7af0491feda46) I will look into it more tomorrow

Thanks! Will give it a go and see.

jkwim, 2023-01-08:
I ran in to some issue when trying to upgrade to latest version.

I was on 1.15.258 and tried to go to 1.15.266 and sort of bricked the device. It get stuck at WiFi connection I think as the LED keeps on flashing. Tried all actions using the button and could not revive.

I observed that the WebApp interface has changed and that there is no longer an option to backup LittleFS or delete LittleFS.

I opened the device again and reflshed the device to 1.14.124 which was my original starting point.

[image]

However this flashing also went wrong back to square one. 

I originally had two files in LittleFS, one test file and autoexec.bat.

Has there been major changes in the OTA process since version 258?

I can try directly going to 276 during flashing. Wonder whether the partitions have changed or something like that beyond 258.

jkwim, 2023-01-08:
Update:
Found the reason.

I managed  to get in to emergency AP mode by doing power cycle x 3 times squence.

Then only I realized that I had added startup command `lfs_append wifi_state` during my progressive upgrades to test the new command. I have used the wrong syntax: `lfs_append filename command`

Probably the versions which did not support the new command `lfs_append` just ignored the command but the version which had the command compiled in failed due to above syntax error and got stuck without connecting to WiFi.

Managed to get out of Emergency Mode by clicking the Exit Safe Mode button shown on the Web GUI and then go and clear the Start Up command.

As the next step I added the startup command `lfs_append wifi.log hello` but rebooting failed as before.

So my conclusion is that `lfs_append` needs to be investigated.

Rebooting was successful afterwards.

Backlog lfs_append wifi.log wifi_state

openshwprojects, 2023-01-23:
@jkwim 
lfs_append will not do anything if used with less than 2 arguments
[image]