openshwprojects/OpenBK7231T_App #567: Color mismatch

WiredLife, 2022-12-27:
As described by others too, there is a big color mismatch.
It makes no difference if the color is transmitted with high or low brightness or as rgb or hsb.
Only the outer ring on the color wheel match.
If i set the saturation on webif of openbk higher, there is the possibility that the color match.
Looks to me its a saturation problem.
[image]

openshwprojects, 2022-12-27:
Have you tried that new exponential colour mode?

WiredLife, 2022-12-27:
Exponential color mode?
I knew the brightness thing.
Where can i enable the color expo?

openshwprojects, 2022-12-27:
Sorry, I meant brightness. Have you tried all the options?

WiredLife, 2022-12-27:
Yes tried all. As i said, it looks like the saturation is the problem

WiredLife, 2023-01-04:
https://github.com/openshwprojects/OpenBK7231T_App/pull/447
finally fixed it

openshwprojects, 2023-01-08:
You have convinced me. We're merging #447 soon. I will allow float values for PWMs. I am closing this issue now, let's continue discussion in #447 .

closed 2023-01-08 by openshwprojects