openshwprojects/OpenBK7231T_App #569: HA mqtt cfg - "brightness_state_topic" missing

Weafyr, 2022-12-29:
**Describe the bug**
After copy yaml cfg to HA i struggled to synchronize dimmer between HA and device, I found that "brightness_stat_topic" is not generated into config. After manually adding topic dimmer value is correctly synchronized as well.

**Firmware:**
- Version 1.15.247
- Device [Solight LED WIFI smart RGB strip for TV, 4x50cm, USB (WM58) ](https://www.solight.cz/en/solight-led-wifi-smart-rgb-strip-for-tv-4x50cm-usb-detail-3I7K000201.aspx)
- Chip/model: BK7231N
- Device config
P6   - PWM - 3
P8   - PWM - 2
P17 - IRRecv
P24 - PWM - 1

**Screenshots**
[image]
expected
[image]

openshwprojects, 2022-12-29:
Thank you, it was missing, can you check now?
https://github.com/openshwprojects/OpenBK7231T_App/commit/354a74ff9f2806316a20e050d541a3497b67725f
I also see that I must update CW and single color templates.

Weafyr, 2022-12-29:
Work like a charm. Thanks!

openshwprojects, 2022-12-29:
It seems that also CW generator is not using LED driver. @iprak , have you forgot about that one? I will push a fix within 10 minutes for that.

@Weafyr no problem, if you find any other suggestions, please post them here (in separate issues)

closed 2022-12-29 by openshwprojects

iprak, 2022-12-29:
@openshwprojects I think I lost track of that issue. I think that was the issue mentioned on elektroda.com forum?

openshwprojects, 2022-12-29:
@iprak yes, you are correct. There was a user who noticed that the 2 PWM device (CW) is added as raw channel access (separate cool and warm) instead of a LED driver access (with a temperature slider and brightness slider)