openshwprojects/OpenBK7231T_App #571: IMPORTANT Don't show WiFi password as plain Text?

LuisThe0ne, 2022-12-30:
I recently noticed that all devices show the WiFi password as plain text.
I think this is a HUGE Security Issue, because as it happended to me:
When the device doesn''t connect to WiFi (idk how that happened) it openes an access point and anybody can connect to get the WiFi password.
I don't think that's good.

closed 2022-12-30 by openshwprojects

openshwprojects, 2022-12-30:
Thank you for your suggestion, I have hidden those fields. Is there anything else I can fix?

LuisThe0ne, 2022-12-30:
Great, thanks!
Wouldn't have expected such a quick fix, but as far as I can see the build failed.
And secondly, the passsowrd is sill stored in plain text in the flash, or?
Thanks

LuisThe0ne, 2022-12-30:
Upgraded already, works great, thanks!
Maybe a bit offtopic, but would you mind taking a look at my PR for the webapp?

openshwprojects, 2022-12-30:
I will take a look at that when I have some more time, but can you provide screenshots what it exactly does?

LuisThe0ne, 2022-12-30:
Screenshots are in the PR

jkwim, 2022-12-31:
Also I once observed that Log exposes WiFi and MQTT passwords as well.

I could not figure out how to reproduce this log though.

```
Info:GEN:PIN_SetupPins pins have been set up.
Info:MAIN:Main_Init_Before_Delay done
Info:MAIN:Main_Init_Delay
Info:MAIN:Main_Init_Delay done
Info:MAIN:Main_Init_After_Delay
Info:MAIN:Using SSID [......]
Info:MAIN:Using Pass [......]
```
```
Info:MQTT:mqtt_userName .....
mqtt_pass ....
mqtt_clientID .....
mqtt_host 19....:1883
```

Please look in to fixing this as well.

wehrmannit, 2023-04-22:
I noticed that the app also stores the wifi password in plain text in the URL - I was a bit confused when I typed in my SSID into the search bar and found my password in clear text.
[image]