openshwprojects/OpenBK7231T_App #573: Feature Request - MQTT Toggle

bgardner46, 2022-12-31:
I use a gui that toggles an output. Until now all of my devices have been flashed with Tasmota and toggle works as an MQTT paylad. It seems like the only payload supported in OpenBK is either 1 or 0. I would like to use for example:

ktchen_light/1/set toggle

Is this currently possible or can it be added?

Thanks.

openshwprojects, 2022-12-31:
Hello, have you tried Tasmota syntax, which is supported?

cmnd/obkDevice/POWER with payload "toggle"
(if multiple relays are present, use POWERX where X is a number)

bgardner46, 2022-12-31:
I have not - didn't realize that the Tasmota syntax was supported. At the
moment I have utilized some programming in my home control system to
accomplish the same thing but I will definitely give it a try tomorrow.

Thanks for the quick reply

Bob

bgardner46, 2023-01-01:
I couldn't wait. I just tried this and it worked perfectly. The more I learn, the more I like this project.

Thank you.

Bob

openshwprojects, 2023-01-02:
Thank you, is there anything else I can help you with?

closed 2023-01-02 by openshwprojects