openshwprojects/OpenBK7231T_App #574: Driver Support for SHT3x

Chasbrot, 2023-01-01:
**Describe the bug**
First Happy New Year! I got a feature request. I have a super generic aliexpress temperature and humidity sensor based. Flashing and GPIOs are working, but the driver for the main sensor SHT30 is missing. I have looked at the supported devices and found two more that have the exact same sensor. 
https://www.elektroda.com/rtvforum/topic3933836.html
https://www.elektroda.com/rtvforum/topic3791330.html

**Firmware:**
- Version [1.15.251]
- Chip/model: BK7231N (CB2)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2023-01-01:
Hey, thanks for the suggestion, that's a great one!

I just bought SHT30 on ebay, lets wait for it to arrive.

Bbut are you aware that one linked device supports SHT30  but through TuyaMCU (so no direct SHT30  support) and the other one is just put on list because we also list non-WiFi devices for the sake of information?

If your device has SHT30  connected to TuyaMCU then it's most likely already supported.

Chasbrot, 2023-01-01:
I just took a quick look at it again, there is no tuyaMCU, only the two ICs. Yes, i have read both devices, but i thought they might all use the same driver. Yes, i will submit the new device. Thanks for the quick response.

catdogmaus, 2023-01-04:
I did manage to flash a very similar temperatute/humidity device with BK7231N like this https://www.elektroda.com/rtvforum/topic3874289-150.html#20181014 - that is not my post but my device is identical. This also does not have a driver, also I have no idea what to do about the pin setup. Eventually, they could have the same setup. It would be great if somebody could help.

superjedi, 2023-01-05:
> I did manage to flash a very similar temperatute/humidity device with BK7231N like this https://www.elektroda.com/rtvforum/topic3874289-150.html#20181014 - that is not my post but my device is [...]

Apparently this device is using a CHT8305 temperature and humidity sensor: http://sensylink.com/a/products/lm2/CHT8305.html

catdogmaus, 2023-01-05:
> Apparently this device is using a CHT8305 temperature and humidity sensor: http://sensylink.com/a/products/lm2/CHT8305.html

Hopefully it will also get support at some point

superjedi, 2023-01-06:
> Hopefully it will also get support at some point

I can probably help, can you please share instructions for flashing with positions of signals ? I've received a bunch of these sensors yesterday.

Chasbrot, 2023-01-06:
> > Hopefully it will also get support at some point
> 
> I can probably help, can you please share instructions for flashing with positions of signals ? I've received a bunch of these sensors yesterday.

@superjedi: flashing instructions are at the bottom of the main github page. The position of the signals (you probably mean pinout) are in the datasheet of your ic, because they vary from model to model. Just google your part number with "datasheet" attached and search a bit.

superjedi, 2023-01-06:
> @superjedi: flashing instructions are at the bottom of the main github page. The position of the signals (you probably mean pinout) are in the datasheet of your ic, because they vary from model to model. Just google your part number with "datasheet" attached and search a bit.

This request was directed to @catdogmaus as he apparently did successfuly flash the same exact PCB I have in hand and that could save me some time needed to remove the RF shield and hassle trying to figure out which pins are which.

As for the instructions I meant, particular instructions relating to this PCB design, as if there are signals/tracks that have to be cut off (such as UART rx/tx that could be shared with some other chips) and/or EN/RST pins that could be used to avoid scratching off tracks.

I'm not looking for help for stuff that has obviously been already well documented, thank you.

openshwprojects, 2023-01-06:
Which pins are used for I2C communication on your devices?

I need to know is it a hardware I2C or a software.

catdogmaus, 2023-01-06:
> I can probably help, can you please share instructions for flashing with positions of signals ? I've received a bunch of these sensors yesterday.

Hi. Unfortunately, OB help files are at the moment like a bottomless rabbit hole. To find the correct information to flash your device could be a real challenge. BKwriter does not work on BK7231N!! They have also that new flashing tool but it seems not to be ready yet. I used python tool and latest BK7231N_QIO file. You don't need to cut any tracks - at least I didn't do anything like that. There are different ways to reset the chip to start flashing. For me, the only way that actually worked was disconnecting for a moment the 3V power to the chip. For soldering, and pin layout BK7231N look here https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076 . When u need more detailed information about flashing in python let me know. I considered creating correct instructions for that or maybe Y-tube video but when they get out their new flashing tool, the python method will be obsolete anyway.

superjedi, 2023-01-06:
> As I already mentioned. This device seems not to have a driver in openbeken jet! At least I receive an error complaining about the driver when I try HA setup page in the device table. That means at the moment u will not be able to use it anyway.

Ok, I'll try to work on a driver for CHT8305 this week-end, let's not hijack this issue any more since it's dedicated to another PCB and another temperature sensor (SHT30).  You can reach me on discord if you want, my handle is superjedi#3820

catdogmaus, 2023-01-06:
> let's not hijack this issue any more since it's dedicated to another PCB and another temperature sensor (SHT30)

Yeah, I already thought that it might be wise to move this topic over to some dedicated to CHT8305 but hopefully, the most burning questions are now resolved.

ByteMangler, 2023-01-08:
CHT8305 activity here: https://github.com/openshwprojects/OpenBK7231T_App/issues/584

Dheenhasty, 2023-01-14:
do a quick and dirty implementation and tested on one of my BK7231N-CBU generic Temp/Hum sensor. 

I have copy what was done for the CHT8305T with correct address base on Datasheet for the SHT3X. 

Just implemented the single shot retrieval from the sensor. with following startup script : 

init:
	echo "Init driver and powersave "
        startDriver TuyaMCU
	PowerSave
again:
        echo "activate SHT3X Sensor"
	startDriver SHT3X
	echo "Send some measure"
	delay_s 5
	echo "stop SHT3X driver"
        stopDriver SHT3X
	echo "wait 10min"
	delay_s 600 
	goto again

And below pin config : 

[image]

[image]

there's a small issue regarding the base ref of the sensor, i think the board generate some heat so temp has a more 4 Celsius degree than normal. but it can be ajusted in Home Assistant on mqtt value retrieval 

now only concern is regarding the consumption.

Dheenhasty, 2023-01-14:
Documentation i rely on : https://www.mouser.com/datasheet/2/682/Sensirion_Humidity_Sensors_SHT3x_Datasheet_digital-971521.pdf

 - Change Start Address based : 44
 - Launch Single Shot Mode no clock strech in low mode (to prevent high consumption)
 - Do a soft reset of the sensor.

Dheenhasty, 2023-01-14:
I really think it can be greatly optimise ....and help will be welcome here :)

ByteMangler, 2023-01-14:
@Dheenhasty why are you concerned about the consumption? Are we referring to power consumption? I would think the SHT30 is very low power compared to the BK7231.

Dheenhasty, 2023-01-14:
Yes, the power cosumption more global on the appliance/sensor so including the BK7231, the generic temp/humidity sensor that rely on sht30 is using a bk7231n-CBU but is powered only by 2 AA battery ..... and after some test the OBK firmware with sht30 driver has consumed all the battery in 6 hours (with a 10min delay between check and powersave activated)

I suppose it's mainly the wifi who consumed, so it could be nice if we can put the BK in deep sleep during idle phase

openshwprojects, 2023-01-19:
It's wifi module. Deep Sleep is required for that kind of devices. @Dheenhasty helped us to get it working on N, we also hope to get it working on T soon.

Dheenhasty, 2023-01-26:
I think this issue can be closed. the SHT3X support work well other issue here is regarding deep sleep. who is clearly another subject.

what do you think ? 

i know that we need to improve CHT8503 and SHT3 support to include them in I2C Soft standard lib. 

(wondering if it will not be easier to access command ReadBytes and WriteBytes on script mode. )