openshwprojects/OpenBK7231T_App #576: MQTT Group Topics

NielsPiersma, 2023-01-01:
I would need MQTT Group Topics to work for provisioning and configuration purposes.
In Tastmota we use "cmnd/tasmotas/"
[image]

I was hoping that in OpenBK this would relate to the Device Groups
[image]

But I can't get multiple devices to respond to commands I target there.

[image]

1) Is my assumption correct that device groups in OpenBK are not the same as Group Topics in Tasmota?
2) Can we have Group Topics in OpenBK?
3) Would the assumed syntax as "cmnd/[group topic]/power" and Payload 1 or 0 be correct?

Thanks
Niels

openshwprojects, 2023-01-01:
Tasmota Device groups are this:
https://tasmota.github.io/docs/Device-Groups/
It has nothing to do with MQTT.

How group topics works? Are they... "listen only"?

So basically you want.... to set a group topic to "Bekens" on multiple devices, and then publish a "cmnd/Bekens/POWER" with "toggle" argument and they all listen and react?

What about replies - should the device publish something on group topic in a reply?

PS: "cmnd/bekendevice/power" with "0" or "1" or "toggle" already works

NielsPiersma, 2023-01-01:
First, thanks for clarifying the difference between them. At least now it makes sense why I couldn’t get it to work.

well group topic 1 (how I use it) works the same as the device topic. But all the devices subscribe to this topic. If I send a command like cmnd/tasmotas/reset 3 they reboot and recalibrate WiFi. I like doing this for general housekeeping and tidying. But the same goes for mass changing the SSID and WiFi password on a regular basis.

if a device published something in the group topic all subscribers will execute the received command. So not something I would expect regular.

so it is just like the normal full topic where the device receives it command from, but has a name that is shared over all devices. One command there is a command for all (subscribers).

I would not post or publish any reply into the group topic.

I can conform the P S: "cmnd/bekendevice/power" with "0" or "1" or "toggle"  work fine. That what I was trying to expand.

Niels

openshwprojects, 2023-01-01:
So basically:
1. add a field "GroupTopic" to gui
2. listen for commands on both cmnd/myBekenDevice1/POWER (etc etc) and /cmnd/myGroup/POWER
3. ... and execute those commands
And that's all?

NielsPiersma, 2023-01-01:
Yes,

Thats it
Niels

openshwprojects, 2023-01-02:
added:
[image]

NielsPiersma, 2023-01-02:
Super, I'll give it a try and let you know.
Niels

NielsPiersma, 2023-01-02:
Well, I tried, but no response (yet). I configured "bekens" as the secondary group topic.

cmnd/obkBF512F1E/Power      Payload 1  == working
cmnd/bekens/Power      Payload 1 == not working.
[image]

Niels

openshwprojects, 2023-01-02:
You must restart MQTT for changes to take place.

NielsPiersma, 2023-01-02:
I can confirm this is working as expected. Excellent!!
Kudos!
Niels

closed 2023-01-02 by NielsPiersma