openshwprojects/OpenBK7231T_App #579: TESTERS NEEDED. Anyone with UART hooked up to BK7231? BK7231T or N, or both? New flash tool

openshwprojects, 2023-01-02:
I am releasing a new flash tool now, can anyone help with testing?
https://github.com/openshwprojects/BK7231GUIFlashTool
See readme on that repository.

HumbleDeer, 2023-01-03:
Works fine.

ByteMangler, 2023-01-06:
tried it today with a CB3S module BK7231N. Worked as advertised.

Perfunctory85, 2023-01-06:
tried it today with a CBU module BK7231N. Doesnt work....  the tool says "failed to set baudrate". Ive tried all baudrate options. anyone can help? Tried two different usb->uart sticks

ByteMangler, 2023-01-06:
I found that the CP2102 usb->serial device I use doesn't like 9216k so I set the baud rate to 115200. 
@Perfunctory85, are you sure your rx/tx wiring is correct? Very easy to get it the wrong way around! If in doubt, swap the wires.

Perfunctory85, 2023-01-06:
> I found that the CP2102 usb->serial device I use doesn't like 9216k so I set the baud rate to 115200. @Perfunctory85, are you sure your rx/tx wiring is correct? Very easy to get it the wrong way around! If in doubt, swap the wires.

Ive tried every baudrate :( And of course, i swap the wires and check it twice... Doesnt work :(

Perfunctory85, 2023-01-06:
[image]
[image]

ByteMangler, 2023-01-06:
@Perfunctory85, the jumper on your usb->serial board is set to 5V. Not sure how the BK7231 reacts to over-voltage on the serial pins. 
How do you reset the BK7231? cycle power or gnd CEN? 
I gnd CEN and sometimes it takes a few tries.

Perfunctory85, 2023-01-06:
Yes, you are right. Now i put it on 3.3V. But on this way, i cant get the bus.  Tried CEN -> GND and Powercyle.

output auf UART is about 3 V. Also if the jumper is connected to 5v

ByteMangler, 2023-01-06:
maybe your device is also using the serial pins and the usb->serial is fighting with it? If that is the case, then it may be necessary to cut a track or desolder the module. Neither option is ideal.

An oscilloscope would be handy to check if there is any fighting.

Perfunctory85, 2023-01-06:
Thats it. Ive desolder the modul an now i can flash it. Thanks. I hope i get it resoldered :D

ByteMangler, 2023-01-06:
@Perfunctory85  good to hear you were successful.

@openshwprojects might be worth noting my tests were done using an iMac with Windows7 64bit running in a virtual machine with Parallels as the hypervisor and a CP2102 usb->serial chip. As previously mentioned, I needed to set the baudrate to 115200.

dvcomp, 2023-01-09:
Hi!

Not working for me. bk7231tools worked with the same environment. WBLC5 module.

crunchtime2k, 2023-03-03:
Works fantastic on CBU module, thanks

roobbb, 2023-03-11:
Hello.

I tried to follow this instruction: https://www.elektroda.com/rtvforum/topic3912748.html#20145042
Creating a backup with BK7231Flasher.exe works fine. But it creates a file which has 2.048KB. BKwriter 1.60 even dumps fine a file with 1.192KB. I thought it should be the same file size in any case.

Flashing unfortunately always fails for me, no matter what baud rate I use:
115200
```
All selected sectors erased!
CheckRespond_FlashWrite4K: bad value returned?
Writing sector 0x00... Writing sector 0 failed!
Writing file data to chip failed.
```
or 921600
```
Writing sector 0x00... ok! CheckRespond_FlashWrite4K: bad value returned?
Writing sector 0x1000... Writing sector 4096 failed!
Writing file data to chip failed.
```
or 1500000
```
Writing sector 0x00... ok! Writing sector 0x1000... ok! CheckRespond_FlashWrite4K: bad value returned?
Writing sector 0x2000... Writing sector 8192 failed!
Writing file data to chip failed.
```
I have no idea what I am missing or doing wrong.
It seems flashing via python or bk_writer1.60.exe even doesn't work. So it might not be a problem of BK7231Flasher.exe, but I thought I should report here the above behavior. 

Kind regards
roobbb

----
[image]

roobbb, 2023-03-12:
OK, I think I have found my mistake: As stated in the 1st post of the forum's thread above, I had no Vcc connected. Reading flash worked fine that way. Now, with Vcc connected to 3.3V flashing works fine too with this tool 😅

regards
roobbb