openshwprojects/OpenBK7231T_App #580: Device in a weird state; can't make many kinds of changes

TurkeyMan, 2023-01-03:
Device is on 1.15.158
If I try and set the device name, it does nothing.
Clicking "Restart" on the front page does nothing.
Going to web app and filesystem, "List filesystem" or "Create file" does nothing.

If I go to Configure Module and make changes there, that actually works for some reason... any guesses how it got in a weird bad state?
I can do an OTA to latest (1.15.275), it works... device rebooted, new version is running, all same problems exist. Half of the software just doesn't work.

It wasn't this way on installation. I think the fault may have coincided with creating a new `autoexec.bat` file... I created the file, but a command to load tuya and ntp, and then saved the file. Instead of restarting, I ran OTA and it reboot into this weird state.

TurkeyMan, 2023-01-03:
I tried accessing the device with a different web browser (edge), and everything works again... so it seems that it's firefox that's gotten into a broken state.
Does the webpage use cookies or any state management?

openshwprojects, 2023-01-03:
Which firefox?  here I have 106.0.5 and it seems to work?

TurkeyMan, 2023-01-04:
I'm on 108.0.1.
I mean, I'm sure the problem's not firefox; I'd just guess I performed a series of operations in such a way that I entered a bad state. It seems associated with the browser and not the server... so I suspected cookies keeping some state?

openshwprojects, 2023-01-04:
As far as I know, there is no cookies used for native interface.

Your issue sounds like you have two devices with the same mac.

Does your mac ends with 00 00 00?

Have you tried changing it to something else?