openshwprojects/OpenBK7231T_App #582: Keep relay state on / after reboot

NielsPiersma, 2023-01-04:
I don't know if it is a bug. But I've found that all my relays that I've flashed and updated so far suffer from being turned off when rebooted, but not turning on again after reboot. 

The normal operation I've experienced until today is the following;
When you update a smart plug, it reboots. At the moment it reboots the relay is pulled down and the light goes off.
Once started again, the relay stays off. The downside is that Home Assistant also "thinks" the relay is still on.

Luckily with help from @iprak  and @openshwprojects  my plugs, on start, no execute the autodiscovery towards home assistant. This updates the status in Home Assistant. 

But there is no joy in updating 25 plugs which I need to power on again. (Times 25). Is there a way to 
1) have the reboot/firmware update "store" the relay state so after reboot it is powering on again?
2) have the plug request the last state from HA so it is put back in the previous state?

Thanks
NIels

openshwprojects, 2023-01-04:
Nothing were change with "power on state", if something is different, then it's your HA. Maybe HA repeated discovery changes something?

Here is the OpenBK start channel values system, here you can set on OpenBK device to remember last relay state:
[image]

NielsPiersma, 2023-01-04:
That's quick :)
Well, I don't think it is HA. I have the same behavior without MQTT setup. 
But your directions are likely to solve my problems. I'll give it a quick try and let you know.
NIels

openshwprojects, 2023-01-04:
@NielsPiersma  please try with MQTT disabled (enter empty MQTT IP)

NielsPiersma, 2023-01-05:
> @NielsPiersma please try with MQTT disabled (enter empty MQTT IP)

Selecting no IP address and pressing submit results in reverting to the previously entered IP address.

I used a non-existing address for testing.

Flags value 5184

Startup values -1

[image]

Still, plug on -> reboot (relay goes off) -> starting,  relay stays off.

Niels

openshwprojects, 2023-01-05:
Huh. I know what happens! I would never expect that, but well....

You have set Flag 12 - [LED] remember LED driver state (RGBCW, enable, brightness, temperature) after reboot and it collides with raw relay states!

What kind of device do you have?

Currently, the assumption is that user enables flag 12 for LEDs with CW, single color, etc strips , but keeps it disabled for relays
If you enable that flag on relay, then device thinks the it's LED and overwrites relay states with led states.

Can you check without flag 12?

I can also consider adding a fix for that for you, but I am not sure, we are trying to preserve flash wearout and don't use much flash config memory.... flash have a limited amount of erase cycles and wears out after... well tens of years, let's hope.

NielsPiersma, 2023-01-05:
@openshwprojects , You are right; Flag 12 conflicts with the -1. When disabling Flag 12 it works as expected. The smart plug restarts, it briefly powers off and powers on immediately.

I am running an LSC smart plug https://www.elektroda.com/rtvforum/viewtopic.php?p=20350582#20350582

LED strip status etc is not important to me. 

NIels

NielsPiersma, 2023-01-05:
And final question, can I set the startvalue to -1 using the cmnd from MQTT.

I can't find it in the available commands. Now I need to login to 35+ plugs and update the setting manually.

https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands.md

https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands-extended.md

Niels

openshwprojects, 2023-01-05:
What do you @NielsPiersma , should I try to fix Relay status save with Flag 12 set? 

I can add a command for that one, no problem

NielsPiersma, 2023-01-05:
> What do you @NielsPiersma , should I try to fix Relay status save with Flag 12 set?
> 
> I can add a command for that one, no problem

You don't need to fix the relay status. I understand Flag 12 and startup value -1 are mutually exclusive.
Adding the command would be appreciated. Let me know if you've fixed it. I'll test it.

I am so happy with the MQTT group, I can now mass update all OpenBekens and tweak them as a group.

Saves a lot of time ;)
NIels

openshwprojects, 2023-01-05:
Ok I will do. Btw, can you post your method on Elektroda?
Or maybe even better... wait until it's finished and then post it a separate article on Elektroda and I will link to it from here. Thanks.

openshwprojects, 2023-01-06:
SetStartValue [ch][val]
added
https://github.com/openshwprojects/OpenBK7231T_App/commit/9f910c5b73b17f6014f380401c97543c131751f5

NielsPiersma, 2023-01-06:
> Ok I will do. Btw, can you post your method on Elektroda? Or maybe even better... wait until it's finished and then post it a separate article on Elektroda and I will link to it from here. Thanks.

https://www.elektroda.com/rtvforum/viewtopic.php?p=20372001#20372001

openshwprojects, 2023-01-06:
That;s perfect, is there anything else I might add for you?

NielsPiersma, 2023-01-07:
No,not today. But I am certain I’ll stumble on something again and submit a request.
Thanks for now.
Niels

closed 2023-01-07 by NielsPiersma