openshwprojects/OpenBK7231T_App #583: Configuring flags via console / MQTT

NielsPiersma, 2023-01-04:
Can the flags be configured via the console or MQTT cmnd?
The recent addition of the MQTT Group enables us to mass configure settings across many devices.
I would, for example, like to enable Flag 6,10 and 12 (Flags (Current value=4160)) for 25 Smart Plugs

Entering "Flags 4160" on the console results in;
Error:CMD:cmd flags NOT found (args 4160)

Having the possibility to update the flags via MQTT significantly improves the manageability of large groups of devices.

Essentially, I would request that all configuration commands be available via MQTT.

NIels

openshwprojects, 2023-01-04:
Hello,
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/commands-extended.md
"SetFlag" command, but I can add a "Flags" command as well if you want, wait

NielsPiersma, 2023-01-04:
Flags, with the corresponding number (enabling multiple options  in one go) would be helpful.
Niels

openshwprojects, 2023-01-04:
Flags command added, altough I am not sure what when we reach 32+ flags, I tested 64 bit sscanf only on Beken and Windows, but it seems to work okay

NielsPiersma, 2023-01-05:
working like a charm.
I can now distribute these setting by executing 

[image]

I am so happy that I can mass configure my OpenBeken devices now.!!!
NIels

NielsPiersma, 2023-01-05:
Thanks for this new functionallity!
Niels

closed 2023-01-05 by NielsPiersma

iprak, 2023-01-06:
I think what we eventually need is a way to save/apply configuration. If a device had to be reset then the saved configuration could be reapplied. The same configuration could also be applied to other devices of same kind (minus the device id).